Kriechi/Cura-DuetRRFPlugin #1: cura 3.2 + rrf plugin

bloood3d, 2018-02-13:
Hi,
I tried to use the plugin with cura 3.2 and as soon as I try to upload to the printer or print it directly from cura, the software just crash unexpectly without throwing any error.

This is the error that I found in the log of cura.

2018-02-13 11:41:42,799 - CRITICAL - [(18012)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-13 11:41:42,801 - CRITICAL - [(18012)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-13 11:41:42,803 - CRITICAL - [(18012)-MainThread] cura.CrashHandler.__init__ [67]:   File "C:\Users\Tom\AppData\Roaming\cura\3.2\plugins\Cura-DuetRRFPlugin\DuetRRFOutputDevice.py", line 155, in onFilenameAccepted
2018-02-13 11:41:42,805 - CRITICAL - [(18012)-MainThread] cura.CrashHandler.__init__ [67]:     gcode = getattr(Application.getInstance().getController().getScene(), "gcode_list")
2018-02-13 11:41:42,807 - CRITICAL - [(18012)-MainThread] cura.CrashHandler.__init__ [67]: AttributeError: 'Scene' object has no attribute 'gcode_list'

Kriechi, 2018-02-13:
Thanks for the bug report!
Looks like the API in Cura 3.2 changed. I will take a look what I have to change.

Kriechi, 2018-02-13:
fixed with a new plugin version: https://github.com/Kriechi/Cura-DuetRRFPlugin/releases/tag/v0.0.5

closed 2018-02-13 by Kriechi

bloood3d, 2018-02-13:
Ty you’re awesome. This plugin is very useful and save me/us a lot of time :)