Kriechi/Cura-DuetRRFPlugin #2: Network error when trying to upload sliced file

x0rtrunks, 2018-02-16:
I installed 0.0.6 in Cura 3.2.1 and configured the plugin for my DuetWifi's address, but when I slice a file and try to upload it, I get an error message in Cura.

There was a network error: 301 protocol "" is unknown.

Kriechi, 2018-02-16:
Did you enter your DuetWifi's address with `http://` in the beginning, or just an IP address?
301 is usually a HTTP redirect, but maybe this is a Qt error code.

Please post parts of your Cura log. On macOS you can find the log under `~/Library/Application\ Support/cura/3.2/cura.log`.

A working plugin will produce log entries like this:
```
2018-02-15 17:43:01,456 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [144]: Filename set to: test.gcode
2018-02-15 17:43:01,458 - DEBUG - [(140735718765376)-MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [66]: 'info/send_slice_info' is turned off.
2018-02-15 17:43:01,461 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [157]: Loading gcode...
2018-02-15 17:43:01,461 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [177]: Connecting...
2018-02-15 17:43:01,554 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onConnected [184]: Connected
2018-02-15 17:43:01,554 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onConnected [185]: Uploading...
2018-02-15 17:43:05,493 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onUploadDone [196]: Upload done
2018-02-15 17:43:05,495 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onReadyToPrint [229]: Ready to print
2018-02-15 17:43:05,518 - DEBUG - [(140735718765376)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onPrintStarted [236]: Print started
```

cbusillo, 2018-02-16:
Cura 2.3.1 crashes on upload/simulate/print for me.  Here is the debug info. 

<code>2018-02-14 10:50:34,205 - INFO - [(140736073368448)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin Cura-DuetRRFPlugin
2018-02-14 11:07:12,403 - DEBUG - [(140736073368448)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [139]: Filename set to: REX2_431c5753-eabe-43b8-bc1a-a47ae5b24fbd.gcode
2018-02-14 11:07:12,435 - DEBUG - [(140736073368448)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [152]: Loading gcode...
2018-02-14 11:07:12,554 - CRITICAL - [(140736073368448)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/cbusillo/Library/Application Support/cura/3.2/plugins/Cura-DuetRRFPlugin/DuetRRFOutputDevice.py", line 155, in onFilenameAccepted</code>

Kriechi, 2018-02-16:
... the most important lines should be right after the last one you posted...

cbusillo, 2018-02-16:
2018-02-14 11:07:12,403 - DEBUG - [(140736073368448)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [139]: Filename set to: REX2_431c5753-eabe-43b8-bc1a-a47ae5b24fbd.gcode
2018-02-14 11:07:12,435 - DEBUG - [(140736073368448)-MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [152]: Loading gcode...
2018-02-14 11:07:12,482 - INFO - [(123145391353856)-Thread-2] SliceInfoPlugin.SliceInfoJob.run [29]: Sending anonymous slice info to [https://stats.ultimaker.com/api/cura]...
2018-02-14 11:07:12,551 - CRITICAL - [(140736073368448)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-14 11:07:12,552 - CRITICAL - [(140736073368448)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-14 11:07:12,554 - CRITICAL - [(140736073368448)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/cbusillo/Library/Application Support/cura/3.2/plugins/Cura-DuetRRFPlugin/DuetRRFOutputDevice.py", line 155, in onFilenameAccepted
2018-02-14 11:07:12,555 - CRITICAL - [(140736073368448)-MainThread] cura.CrashHandler.__init__ [67]:     gcode = getattr(Application.getInstance().getController().getScene(), "gcode_list")
2018-02-14 11:07:12,556 - CRITICAL - [(140736073368448)-MainThread] cura.CrashHandler.__init__ [67]: AttributeError: 'Scene' object has no attribute 'gcode_list'

Kriechi, 2018-02-16:
Thanks - this now shows the error clearly.

I believe that this was fixed with v0.0.5 - please make sure to re-download and install the latest plugin release from https://github.com/Kriechi/Cura-DuetRRFPlugin/releases.

closed 2018-02-16 by Kriechi

x0rtrunks, 2018-02-16:
My error was different. It didn't crash.

I tried... 
10.10.0.63
http://10.10.0.63
http://z-bot
http://z-bot.local

all produced the same error.

Today I removed the printer from the plugin and added a new one using http://z-bot.local and it now connects.

Love the plugin. Glad it's working now.