Kriechi/Cura-DuetRRFPlugin #3: Installation Windows and connecting.

dintid, 2018-02-19:
Firstly a thank you for your work :)

In windows 10 it needs to go to:
`C:\Users\<Username>\AppData\Roaming\cura\3.2\plugins\Cura-DuetRRFPlugin`
If **plugins** folder doesn't exist you need to create it.

**Running from source:**
How do you do that?

**Connecting?**
I've setup my url (I'm using local domain) and tried with IP, but I can't connect.
Also tried with and without the _reprap_ password - I don't use any password.

Kriechi, 2018-02-19:
Please post your log file, or at least the parts with `Cura-DuetRRFPlugin.DuetRRFOutputDevice`. 
You can find the log file in `C:\Users\<Username>\AppData\Roaming\cura\3.2\cura.log`.

dintid, 2018-02-22:
A got a bit sidetracked. Turned out it only wanted to take IP addresses and not UNC or URLs.
I also had to put in the IP and the restart Cura. Seems it might load the settings on startup?
You still need log file(s)?

Kriechi, 2018-02-23:
UNC will probably not work. But I think regular URLs should.
I will give it a try with a fresh & clean Cura profile.

dintid, 2018-02-23:
Reinstalled windows and cura and it only works with straight up IPs here.

Kriechi, 2018-02-23:
Ok. I'm running macOS, and I'm using http URLs.
Please provide me with the log file or the parts with `Cura-DuetRRFPlugin.DuetRRFOutputDevice`.

Kriechi, 2018-03-01:
fixed with v0.0.9.

closed 2018-03-01 by Kriechi

dintid, 2018-03-01:
Sorry I didn’t provide logs. I had formatted my machine, but then it suddenly started working on its own?