Kriechi/Cura-DuetRRFPlugin #7: Crash when trying to upload to RRF SD (Cura master branch)

smartavionics, 2018-03-20:
Just tried to use your plugin with the Cura master branch (as of 20/3/18) and got this...

````
2018-03-20 16:42:41,878 - DEBUG - [MainThread] Cura-DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [156]: Loading gcode...
2018-03-20 16:42:41,883 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-03-20 16:42:41,884 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-03-20 16:42:41,884 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/home/markb/.local/share/cura/master/plugins/Cura-DuetRRFPlugin/DuetRRFOutputDevice.py", line 159, in onFilenameAccepted
2018-03-20 16:42:41,884 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     active_build_plate_id = Application.getInstance().getBuildPlateModel().activeBuildPlate
2018-03-20 16:42:41,884 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'BuildPlateModel' object has no attribute 'activeBuildPlate'
````
As you know Cura master is changing all the time (that's it's job!) so they have probably made some incompatible change.

Kriechi, 2018-03-20:
again.... They already broke it from Cura 3.1 to 3.2: https://github.com/Kriechi/Cura-DuetRRFPlugin/commit/620edb35878f63c4666d6ea0413a919284503001

Unless one of the Cura devs wants to provide a fix / workaround / solution, I probably have to wait for @markwal, so I can steal his fix again 😄

smartavionics, 2018-03-20:
Yeah, they break it all the time. No matter, I haven't used the plugin up till now and so can live without it!

Kriechi, 2018-03-20:
I would recommend opening an issue with the main Cura repo, maybe one of the devs can fix it or provide an easy example how to use their API without requiring a rewrite every 2 weeks.

smartavionics, 2018-03-20:
I think they are getting bored with me posting an issue every time I get a crash in the front end because of buggy master branch code. It's permanently in a state of flux. They are currently working flat out on some new products which are coming up for release so I won't bother them with this.

markwal, 2018-03-20:
Looks like they changed getBuildPlateModel to getMultiBuildPlateModel in https://github.com/Ultimaker/Cura/commit/495fc8bbd705f5145fe8312207b3f048a7dcc106

Rant aside: My whole career, the words "Refactor" in a log comment strike terror in my heart when it is into a component that I depend on.  Especially when they say WIP: Refactor because that means that they are chomping at the bit to break you again.  But I've noticed, that "refactor" is basically most engineers favorite things to do.  Especially when they complain about having to do it.  :-)

markwal, 2018-03-20:
Of course, if you change it now, the plugin will only work with master and won't work with 3.2 any more.

Kriechi, 2018-03-20:
For future reference, we should probably look at https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/PostProcessingPlugin.py
and  
https://github.com/fieldOfView/OctoPrintPlugin/blob/master/OctoPrintOutputDevice.py#L294 
after the next release to figure out how the latest API should be used like.

Kriechi, 2018-03-27:
The fix is probably like this: https://github.com/Ultimaker/Cura/commit/495fc8bbd705f5145fe8312207b3f048a7dcc106#diff-a4cb192cad5ce77939fdd0bf0600208d

Let's give the new Cura beta a few days before I make this change. PRs are welcome though!

Kriechi, 2018-03-28:
fixed by #6de6fd8720484317ece782e92297530ba4449665 and v0.0.13.

closed 2018-03-28 by Kriechi