Kriechi/Cura-DuetRRFPlugin #8: Feature Request: change DuetRRF Connection on machine change

cbusillo, 2018-03-30:
Hi!  Is it possible to have the duet connection change when the printer is changed?  I'm running into the issue where I click print and forget to change the output.  Maybe a way to connect the connection to the machine?

Kriechi, 2018-03-31:
I'm sorry - I don't fully understand what you want to change?
Can you explain what you mean by "duet connection change when the printer is changed"?

cbusillo, 2018-03-31:
Sorry this is really hard for me to explain.  I took screen shots that might explain it better.  Basically I want the Duet Connection tied to a particular machine in Cura.
[image]
[image]

Kriechi, 2018-03-31:
Ok - I think I get the idea of what you want.
I need to find time to look into the Cura plugin system - as there is no proper documentation this might be either really difficult to implement - or it might be really easy. 

If anyone wants to contribute and send a PR that implements this feature - I'm happy to merge it!

GT1ZA, 2020-05-13:
Hi, im using cura 4.5 and cant seem to find this in either the marketplace or get the extension to appear with manual installation. Is it working in 4.5?

Kriechi, 2020-05-13:
@GT1ZA please open a new ticket in the future for new questions and problems.
The latest Cura version 4.6 is supported by the Cura-DuetRRF plugin.

Kriechi, 2020-08-29:
@cbusillo I just pushed a new branch with a complete refactoring of the plugin settings - and I think this should address your feature request of having the settings attached / switched with the Cura printers.

Are you comfortable installing the plugin "from source" into Cura from this branch?
https://github.com/Kriechi/Cura-DuetRRFPlugin/tree/new-settings
Basically download the [zip archive](https://github.com/Kriechi/Cura-DuetRRFPlugin/archive/new-settings.zip) and extract it into `~/Library/Application Support/Cura/4.7/plugins/`

cbusillo, 2020-08-29:
Hi! I no longer have the printers. I think others will appreciate this feature though! Would you like me to close this?

ericlmccormick, 2020-10-09:
This is greatly appreciated. Thank you!

Kriechi, 2020-10-09:
@ericlmccormick did you test it? can you report it is working an does what you expected it to do?
I'm not sure yet how to handle existing user data and printer configurations. Migrating them over to the new settings and matching them against Cura printers is difficult and error prone...

ericlmccormick, 2020-10-09:
Yes. I have two printers running duet boards. I had been using PrusaSlicer for a while. I think when I last used Cura I was printing via octaprint. I noticed that my start and end codes were not working using the plug in from the cura market. I think your plug in sent the code generated, ignoring the machine setup into which includes the start and stop gcode. I might be wrong and may have messed something up.
However, I manually installed the plug-in from here and was able to link the machine profiles to their duet addresses similar to the octoprint plugin. Now when I select a machine, it grabs the correct profiles and sends to the correct machine. This really helps the work flow and grabbed the start and end gcode commands.

Kriechi, 2020-10-10:
I've merged my branch with the new settings integration and will let it sit there for a bit so people can test it, before I release a new version to the Cura Marketplace.

commit ca03f698 (2020-10-10): use Cura printer metadata for storing config

fixes #8
closed 2020-10-10 by Kriechi