Kriechi/Cura-DuetRRFPlugin #10: Fails when there are special characters in filename (e.g. "(")

zylantha, 2018-04-01:
When the uploaded filename to the Duet includes a "(" character, the plugin appears to fail, with messages on the Duet console such as:
Error: M32: GCode file "/gcodes/KPRO_Shiny Amur .gcode" not found

(In this instance, it was "KPRO_Shiny Amur (1).gcode" that I uploaded - it truncated the filename at  the open bracket).

Perhaps there needs to be an escaping of the filename - is it uploading over HTTP? In this case I think the filenames would need to be escaped?

Kriechi, 2018-04-01:
Parenthesis are interpreted as "comments": https://www.duet3d.com/forum/thread.php?id=4723#p41212

I think we need to use quotes around filenames.

beikeland, 2019-04-14:
I accidentally uploaded a file with a question mark in it, maybe less likely to be an issue as it only happened because I change keyboard layouts and they have _ and ? on the same key, so sometimes i get the wrong one.

Upload got to 2% and got stuck, verified with another attempt, same outcome. Same result with both print and upload buttons.

Kriechi, 2019-04-15:
I don't think `?` is an allowed character for filenames in general - but yes, ideally my plugin would display an error message.

beikeland, 2019-04-15:
yes and no, linux escapes them
```
$ touch test?file
$ ls
'test?file'
$ rm test\?file
$ ls
$
```

Kriechi, 2020-01-04:
fixed by 84ff2e13d4007247b07ce20b8a12a9cb64d9ee2f.

commit 33b063cb (2018-04-01): fix filename escaping

fixes #10
closed 2020-01-04 by Kriechi