Kriechi/Cura-DuetRRFPlugin #12: Win10_x64, RRF-Plugin-Activation field is not active

bmwbiker69, 2018-06-28:
Hello,
I'm using win10_x64 with Cura 3.4 (same problem in 3.3 and 3.2)
When I want to activate your Plugin I can open the pulldown-list an see the DUET-RRF.
When I'm moving with my mouse-cursor onto the Field (DUET-RRF) the next field opens
(DUETRRF Configuration).
[image]
But this field is not active!
I can move the cursor onto this field but there was no feedback by changing the colour to blue and the field is "only text"
I can't activate the Configuration-Window?

I've test the same procedure in Win7_X64 and - I can open the Configuration-Window.

Pleas help me.

Thanks in advance,
bmwbiker

Kriechi, 2018-06-28:
Please post your log file, or at least the parts with `Cura-DuetRRFPlugin.DuetRRFOutputDevice`.
You can find the log file in C:\Users\<Username>\AppData\Roaming\cura\3.3\cura.log.

Also make sure to have the latest plugin version installed: https://github.com/Kriechi/Cura-DuetRRFPlugin/releases/latest

bmwbiker69, 2018-06-29:
Dear Thomas,
I found out that - if the RRF-Plugin is installed - all Items in "Erweiterungen" have no respose to mouse hover over but if I use the Keyboard to navigate in Cura I can open DuetRRF-Configuration and set all values to my Printer.

This only happens when DuetRRF-Plugin is installed!
If I remove the RRF-Plugin the "Erweiterungen"-Dropdown menue responses to Mouse hovering.

Thanks in advance,
Klaus
[image]
[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/2150985/cura.log)