Kriechi/Cura-DuetRRFPlugin #13: Publish in Cura Toolbox

ChrisTerBeke, 2018-07-24:
Hi!

Did you know you can publish your plugins in Cura's official Toolbox? This will allow Cura users to discover your plugin more easily as well as install it with the click of a button!

You can find some details about that on https://github.com/Ultimaker/Cura/wiki/Plugin-Directory. Feel free to contact us at plugins@ultimaker.com if you have any questions.

Chris - Ultimaker

johnkray, 2018-07-24:
I second this! I would love to see this plugin in Cura's official Toolbox.

Kriechi, 2018-07-26:
This is in progress... stay tuned!

3D3LTA, 2018-10-04:
Hi. I tried using the plugin with Cura 3.5 and did not work.

I also noticed that it doesn't appear on the toolbox anymore.

ChrisTerBeke, 2018-10-04:
The plugin needs to be updated and re-tested with Cura 3.5, then submitted to us before it appears again. It's the task of the plugin author to do this.

Kriechi, 2018-10-05:
A constant whack-a-mole with Cura updates and breaking API changes...
I haven't seen a single Cura v3 release that doesn't break my plugin.
I guess I'm just not reading their docs properly (or I use too many private APIs).

I'll try to fix it in the next days.

Appesteijn, 2018-10-05:
Sorry for that @Kriechi we are trying to set up a better API system so this won't be needed in the future. If you want to have automate creating a package, you can create a contributor account here: https://contribute.ultimaker.com and use it to send us your package.

Kriechi, 2018-10-05:
Slightly unrelated question: @Appesteijn: I guess your own examples aren't updated yet either? https://github.com/Ultimaker/UraniumExampleOutputDevicePlugin

I guess this would be the most relevant example for my plugin - I'm also looking at the Octoprint plugin, and try to copy their recent changes.

Appesteijn, 2018-10-05:
Yes you are right, a good, up to date plugin would be: https://github.com/fieldOfView/OctoPrintPlugin

Kriechi, 2018-10-05:
@ChrisTerBeke ... in review ... let me know if you need anything else.

ChrisTerBeke, 2018-10-05:
@Kriechi awesome! Cura team will do a quick test on Monday and then approve it. You'll receive an e-mail automatically after which you can hit publish at any time.

Kriechi, 2018-10-05:
*grmpf* yeah - take it slowly. 
Looks like the API changed more than expected and I have to rewrite a few more parts, and release a new version...

Kriechi, 2018-10-06:
Done and published! Thanks!

closed 2018-10-06 by Kriechi

Kriechi, 2018-10-08:
@ChrisTerBeke: not sure where to report / discuss this - so I'll just highjack this thread:

If I look at my now published plugin in the Cura Toolbox, I see this description text: `Provides direct upload of gcode to DuetRRF.` -- I'm not sure where you got this text, or somebody at Ultimaker write it?
It would be nice if this text is taken from the readme, or some other user-editable field in the web-app.

Also, it would be nice if the web-app to upload and manage the plugins would read the plugin version information from the `plugin.json` that I upload. I don't see a reason to enter it manually.

ChrisTerBeke, 2018-10-08:
The description text is taken from the description input field. But it might have been incorrectly copied (at the moment we're copying the content from the portal to our 'old' release system until we switch over the back-end somewhere in the coming weeks. After that it automatically uses the data you entered).

As for getting the data from plugin.json, we're actually planning on phasing out plugin.json (as we now have package.json internally) for package related data, it's just still there so Cura can function properly.

fieldOfView, 2018-10-08:
> As for getting the data from plugin.json, we're actually planning on phasing out plugin.json (as we now have package.json internally) for package related data, it's just still there so Cura can function properly.

So then developers have to edit package.json files again locally, and the idea of the portal was that the portal creates those. Right?

ChrisTerBeke, 2018-10-08:
No, package.json will always be auto-generated based on the metadata you enter in the portal (or post via the API). But we're still figuring out how to deal with plugin.json fields like the i18n, so for now we'll have 2 json files (package.json being generic and plugin.json being specific for plugins).

Kriechi, 2019-01-19:
@ChrisTerBeke sorry to resurrect this thread: Is there anything going on with the review process? I'm trying to publish a new version of this package, but I haven't heard anything from the review team for a few weeks now...

ChrisTerBeke, 2019-01-19:
That should not happen. I’ll ask them what’s up on Monday. Thanks for the heads up.