Kriechi/Cura-DuetRRFPlugin #14: Print Settings are truncated on Upload

wilriker, 2018-08-10:
When I upload GCode to my Duet Wifi with this plugin the settings that Cura appends at the end of the file will not be saved to the SD card. The file will end with
```
; End of Gcode
```
If I save the same Gcode in Cura first to my lokal disk and upload it from there the settings will be present.

I sometimes need to check with which settings I sliced something and only by chance that this information was missing - of course in a case where I would have needed this info. :-(

Kriechi, 2018-08-10:
Thanks @wilriker! I think I fixed it - please try the latest version of the plugin and let me know if it works.

closed 2018-08-10 by Kriechi

wilriker, 2018-08-12:
I can confirm this as fixed! Thanks! :+1:

commit ecba95c3 (2018-08-10): fix gcode serialization

fixes #14