Kriechi/Cura-DuetRRFPlugin #15: Failed to delete file

johnkray, 2018-12-18:
I get an error message every time I upload a file to my printer. The file always uploads just fine and prints fine, so I'm not really sure what the message is about. The error message is below. 

Error: Failed to delete file 0:/gcodes/HRN_Nozzle Wiper Pad.gcode

Not sure if this error is with the plugin of RRF.

Kriechi, 2018-12-18:
Duplicate of https://forum.duet3d.com/topic/8194/cura-duet-reprap-firmware-integration-question

I'll fix this in the next days with a new Cura-DuetRRFPlugin version.

johnkray, 2018-12-18:
Ahh didn't see that one. Great, thank you!

Kriechi, 2018-12-19:
closed via https://github.com/Kriechi/Cura-DuetRRFPlugin/commit/0c2f003cc4c7a98cb58064a01ca5be0c065fb638

v1.0.1 should be available soon.

closed 2018-12-19 by Kriechi

Kriechi, 2018-12-19:
v1.0.1 is now published and available in the Cura Marketplace for Cura 3.6 and the 4.0-beta.