Kriechi/Cura-DuetRRFPlugin #19: Question: Uploading Firmware via Plugin

roseg0ld, 2019-05-31:
Do you think it would be possible/feasible to upload the firmware to a Duet via this plugin? I am including a modified version of this with a plugin for my printer (https://github.com/HydraResearchLLC/Nautilus-Cura-Plugin) and it would be awesome if we could include the latest firmware and allow users to upload new firmware from within Cura. With a little bit of guidance I could write it myself, this could become a useful feature for other printers that use the duet as well.

Kriechi, 2019-06-01:
Usually new firmware releases also require modifications to the config files / macros / etc.
So, I would be hesitant to "just upload" and hope for the best.
At least for me, I still prefer looking at DWC to monitor and interact with the printer - after uploading files to it.

All the API endpoints to upload & flash the firmware are there - if you think this is valuable feature, feel free to implement it - and even send a PR - I'm happy to take a look and merge it, given a clear warning message about potential config mismatches.

roseg0ld, 2019-06-03:
I'm hoping to create the functionality to upload firmware, configuration, macros, etc. all at once in a single zip file. Is this a feature you'd be willing to assist with at all? I'm happy to write the code, but I will likely have a few questions about interfacing with the Duet from Cura.

Kriechi, 2019-06-07:
Unfortunately my time to provide assistence is limited at the moment - the only thing you need to interface with the Duet should be documented here https://github.com/chrishamm/DuetWebControl/blob/master/README.md#communication-to-the-firmware

closed 2019-09-07 by Kriechi