Kriechi/Cura-DuetRRFPlugin #20: Trying to understand plugin operation

Jens5395, 2019-06-20:
This isn't really a feature request or bug report but rather me trying to understand some oddities.
I am coming from an environment where Cura was used as a slicer for a Creality CR10 Marlin based printer and I am trying to set up an additional Duet based printer.
Under Marlin,  to add a printer, I would start Cura, go to settings and add a networked printer - all is well. For my Duet based printer, Cura doesn't recognize that there is an available network printer nor does it talk to it when I specifically enter the IP number of this new printer. Instead, I have to set up the new printer as a locally attached printer.
There may be a reason for this different operation and I would like to understand why this has been set up this way. Alternatively, maybe I am doing something wrong in the setup ?
Also, when I select a printer, slice my model and then want to print this model, Why do I need to select between printing it to different printers - slicing is done for a particular printer so I am not seeing the reason to offer the user the option of printing to a printer that is potentially incompatible with the g code just generated.
Thanks!

Jens5395, 2019-06-20:
Never mind about the second part of my question .... it seems that the first option that comes up after slicing is printing on the printer that the model was sliced for. I was apparently getting confused.
I would still like to figure out why Ihave to select the Duet as a local printer rather than a networked printer.

Kriechi, 2019-06-26:
If I remember correctly, "networked printers" was not yet a thing in Cura, back when this plugin started. If it was, it was only used by Ultimaker themselfs.

The current integration of this plugin with Cura and Duet RRF is very simple and just sends a few HTTP commands. Do you see any major advantage of changing it to a "network printer"? I havce never used this type of integration - but I'm happy to take feedback and pointers on what you are looking for.

Kriechi, 2019-11-10:
I added a note to the README to explain this a bit more.

closed 2019-11-10 by Kriechi