Kriechi/Cura-DuetRRFPlugin #21: Connecting to Duet through Cura with your extension

Bwelford, 2019-09-05:
I have been trying for a couple of days to connect my duet2 wifi control board with cura 4.2.1 and this extension.  I have been looking at my network and have made many changes to improve the success of making a connection, running windows 10, cura.exe, curacli.exe, and curaengine.exe in windows 8 compatibility mode to get cura to run on my windows 10.  I am at my wits end.  I can connect to the printer via chrome and the ip address of the printer.  I have added the printer via the extension and still no luck; cura does not list the printer in the manage printer list.  I have tried to add the printer in cura through the ip address and also cura responds with no connection.  I am attaching the cura log that you ask many people to send.  Let me know what else I need to convey to you.  Thank you for your assistance.  Bruce
[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/3577209/cura.log)

Bwelford, 2019-09-05:
I have no line M551 in my config.g

roseg0ld, 2019-09-05:
You have to add the printer from the non networked printers menu and then upload/simulate/print options will appear after clicking slice

Bwelford, 2019-09-05:
OMG you have got to be kidding me!!  My printer is not listed in the non networked printers, so I had picked a custom Deltabot(which I did build a delta).  And as you have suggested, upload the stl, slice, and presto, my printer appears in the blue box!  Well thank you very much for the quick response.  I expected the printer to appear in the manage printer list, so that is why I never tried to upload a file etc.  Again, thank you.....

Kriechi, 2019-09-07:
Sorry for the confusion - a few more details in https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/20#issuecomment-506086060

closed 2019-09-07 by Kriechi