Kriechi/Cura-DuetRRFPlugin #22: Invalid filename characters causes internal error on G32

moozphat, 2020-03-23:
I have noticed that space characters (%20) in the file name causes internal errors when the plugin sends the G32 command. The file is uploaded properly on the server but has to be started manually through the server. Not sure if this is issue is really a plugin or a DWC2 issue. 

If it is a DWC2 issue (which I doubt since I can start through the server), some processing on the filename could be added to resolve this.

BTW, thanks for the great plugin!

Kriechi, 2020-03-25:
Hi @moozphat - thanks for the report. Which Cura version and which plugin version are you using?

Two years ago, I fixed a similar problem - and I expected it to take care of spaces as well: 33b063cb3bf61a58dd7801389944246429dded53

It might be a few days before I can connect to test printer connected again.

moozphat, 2020-03-25:
This was with Cura 4.5.0 and plugin 1.0.4. I kept getting this error: "There was a network error: 401 Error transferring http://PrinterIP/rr_gcode?M32+%220[...]"

Kriechi, 2020-04-04:
I just submitted v1.0.6 into the Cura Marketplace for review - it should address this issue.
If you want, you can install the latest release manually and test it - or just wait a few days for the release to be available through the Marketplace.

commit 8c6eeb44 (2020-04-04): correctly encode special characters in filenames

fixes #22
closed 2020-04-04 by Kriechi