Kriechi/Cura-DuetRRFPlugin #24: cura plugin crashes after invoking print to ..

elsbiet, 2020-05-06:
the printer starts printing but cura died

[image]

Kriechi, 2020-05-06:
Thanks for the report!
I believe I fixed this (or a very similar) problem in on of the recent bugfix releases - which are available on GitHub, but haven't made it yet through the Ultimaker Marketplace review process.

Please provide your Cura log file, so I can take a closer look.
You can find your log file somewhere around here:
  * Windows: `C:\Users\<username>\AppData\Roaming\cura\4.6\cura.log`
  * macOS: `~/Library/Application Support/Cura/4.6/cura.log`
  * Linux: `/home/<username>/.local/share/cura/4.6/cura.log`

Kriechi, 2020-05-06:
Are you sure you are running plugin version 1.0.7?
https://github.com/Kriechi/Cura-DuetRRFPlugin/blob/v1.0.7/DuetRRFOutputDevice.py#L203 is a different line number than your screenshot...

I think you might be running a more recent or even a custom version?

elsbiet, 2020-05-07:
you're right - it's 1.0.8 (from github) - 1.0.7 seems to be incompatible with duet3 webserver(-*) running on rpi.

attached the relevant part of cura.log. the crash is not really good reproducible because it does does not happen after each invocation of 'print on ..'

[cura-crash.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/4591206/cura-crash.log)

Kriechi, 2020-05-07:
v1.0.9 is now available in the Marketplace - it support the Duet3+SBC.
I believe this is fixed - if not, please reopen with a new crashlog of v1.0.9

elsbiet, 2020-05-17:
no change
[cura-crash-part.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/4640050/cura-crash-part.log)
..and cura dies on subsequent user interaction.

Kriechi, 2020-05-17:
@elsbiet I just pushed a potential fix to the master branch - would you mind following the "Running from source" instructions: https://github.com/Kriechi/Cura-DuetRRFPlugin#running-from-source

Try it and report back with the log again please - even if it works - I'd still like to see the log to understand this behaviour. Thanks!

elsbiet, 2020-05-17:
i will try it and report when my current print will have been finisched - estimated 29+ hours from now.

elsbiet, 2020-05-19:
after the 4th successive 'Print on ..' cura was termitaed by a  segmentation fault - giving up.

these are the last lines from cura.log

2020-05-19 08:06:50,192 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [202]: tgo-print | rr_connect failed with errorCode 203
2020-05-19 08:06:50,192 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [204]: tgo-print | errorCode indicates Duet3+SBC - let's try the DuetSoftwareFramework API instead...
2020-05-19 08:06:50,277 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: tgo-print | Uploading...
c

closed 2020-05-19 by elsbiet

Kriechi, 2020-05-19:
Sorry, but did you close this issue by mistake?
Did this fix your problem or do you still get a segmentation fault error?

elsbiet, 2020-05-21:
> Did this fix your problem or do you still get a segmentation fault error?
things are worse now. segmentation fault is a new behaviour. i've never seen it in cura before.

Kriechi, 2020-05-22:
Do you have more log lines? There should be a "Traceback" near the end, or in the UI pop-up window.

elsbiet, 2020-05-25:
cura crashed because of a segmentation fault - no traceback, no pop-up widow - it just crashed!

Kriechi, 2020-05-25:
Is it reproducible - does it always fail at the same step?
Does the cura log contain anything - even seemingly unrelated to the DuetRRF plugin?

kuon, 2020-12-17:
I have what I think might be the same issue, if I try to print:

```
2020-12-17 17:01:03,329 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: lili-print | Filename set to: CFFFP_Main Pillar - Main Pillar.gcode
2020-12-17 17:01:03,329 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: lili-print | Loading gcode...
2020-12-17 17:01:03,350 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: lili-print | Connecting...
2020-12-17 17:01:03,369 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [202]: lili-print | rr_connect failed with errorCode 203
2020-12-17 17:01:03,369 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [204]: lili-print | errorCode indicates Duet3+SBC - let's try the DuetSoftwareFramework API instead...
2020-12-17 17:01:03,499 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: lili-print | Uploading...
Fatal Python error: Segmentation fault

Thread 0x00007fe2b1ffb640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 888 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe2b27fc640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 888 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe2b37fe640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 2432 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1726 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe2b3fff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.9/threading.py", line 888 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe2d0ff9640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1476 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe2d17fa640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1350 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe30affd640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe30b7fe640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe30bfff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
[... 25 lines cut ...]
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe3127fc640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Thread 0x00007fe312ffd640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/AutoDetectBaudJob.py", line 85 in run
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 132 in run
  File "/usr/lib/python3.9/threading.py", line 950 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 908 in _bootstrap

Current thread 0x00007fe3b204a740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 844 in run
  File "/usr/bin/cura", line 234 in <module>
fish: “cura” terminated by signal SIGSEGV (Address boundary error)

```

cura 4.7.1 with plugin 1.0.11

Kriechi, 2020-12-17:
Does it still happen with the latest Cura 4.8?
Please open this issue with the main Cura project - Segfaults and Python crashes indicate a bigger problem...

kuon, 2020-12-17:
If I use the app image it works, so I guess there must be a problem in my distribution package.

kuon, 2020-12-31:
Apparently the plugin is not using the proper QT API: https://github.com/Ultimaker/Cura/issues/9010