Kriechi/Cura-DuetRRFPlugin #26: Support for a (relative) path in the filename

fieldOfView, 2020-06-26:
If the `file_name` attribute of the `requestWrite` method were to include a path, would that work with your plugin? Does reprap firmware even support uploading to a folder?

I'm working on a plugin that allows the user to specify a folder to save to. This only makes sense for certain output devices; currently only the RemovableDrive and OctoPrint output devices are whitelisted to support this. But if it could make sense to upload to a subfolder via your plugin, I would love to support that.

Kriechi, 2020-06-26:
Yes, uploading to subfolder is already supported, tested, and verified.

I just entered `foo/bar/test.gcode` into the text field of `UploadFilename.qml` and started a full upload+simulate, which as expected created the two nested folders `foo/bar/` and the gcode file `test.gcode` inside it. It the successfully started a simulated print job on this file.

fieldOfView, 2020-06-29:
Ah, I did not realise your plugin comes with a dialog to enter the filename.

Here is a [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/CustomJobPrefix/FYpcfm4TiPBvM9MalbKhx8xChRWOkxXApUXx9HpFQUl4/CustomJobPrefix-v7.0.0-2020-06-29T14%3A59%3A15Z.curapackage?Expires=1908802758&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=Wtgb0Cwgd1eQwrtuTOTkJ26HdF2HTGLWXJ6yQfhAn7wz6fNoky0DcYsIo3euaPL1x2axi8Bgq4JweN4ZxafBHWV6V%2Bp1z%2BWpCbRGQGtX4hvUHaneg5G01UHY9KqGBXXwvy28Buhxyl0wbMij9%2FL4mD8IaL9jR%2BRWZrRIO5LhwGtcKlQ5jK0Z6kA5WLwWTmeF%2FHCCDplNdGpJXPMriocIJ8xuP1q2kMeGkzx7vPQeMpSyky0k7hzRLXR0TFqwn9f7rUBb6GyTpsSG%2FKnuHpG63fFRWzJXKXYvax9Lbw0x9Sp2mpvIXl861z4FSfT%2FAbtFcbzgD2vVqSdEIR1cI7vYKg%3D%3D) of my plugin, which lets the user set a path (and pre- and post-fix) to the filename with a number of replacement patterns. It would be nice to know if it works properly with your plugin (and I don't have a Duet-based printer to test with).

Kriechi, 2020-06-29:
Yes - works good enough.
I'll open a few bug tickets over in your repo.

closed 2020-06-29 by Kriechi