Kriechi/Cura-DuetRRFPlugin #27: Latest commit still crashes Cura.

brianmichalk1, 2020-08-03:
The second print will crash Cura 4.6.1.  Using latest Git plugin.  Running Artemis and Rostock Max V3.

How do I put Python into debug mode for testing a plugin?

Kriechi, 2020-08-03:
Thanks for the report - I tried to reproduce it by simulating and printing a few times but was not able to trigger a crash.

Unfortunately there is no real *debug mode*. Cura itself runs the Python code and interfaces with Qt and the OS.

I've just create a bug report template with additional instructions which would be really useful for debugging - would you mind  just commenting the [requested information](https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/new?assignees=&labels=&template=bug_report.md&title=) into this issue?

broncosis, 2020-08-10:
I also see this 
with the snap install on linux mint 19 on version 4.6.2 it is 100% of the time
and on the appimage of 4.6.1 its about 50% of the time

zlowred, 2020-08-10:
+1 to the issue

**Describe the bug**
Crash a few secomds after the job has been sent to Duet WiFi.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any STL file in Cura
2. Slice it
3. Start printing on the printer built with Duet board by using the plugin's "Print on ..." button, using drop-down to select "Print" option if necessary
4. Observe cura crashed a few secods after the job sent to the printer

**Expected behavior**
Cura does not crash, otherwise everything is the same

**Version Information**
 - Operating System: Mac OS 10.15.5
 - Cura Version: 4.6.1
 - Cura-DuetRRFPlugin Version and Installation: 1.0.10 installed via Marketplace
 - Duet RepRapFirmware Version: 
Board: Duet 2 WiFi (2WiFi)
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.1.1 (2020-05-19b2)
Duet WiFi Server Version: 1.23

** Cura Log**
```
2020-08-10 15:27:32,006 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-08-10 15:27:32,011 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-08-10 15:27:32,015 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin AMFReader
2020-08-10 15:27:32,020 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2020-08-10 15:27:32,026 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-08-10 15:27:32,032 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-08-10 15:27:32,037 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-08-10 15:27:32,042 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-08-10 15:27:32,049 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-08-10 15:27:32,056 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin DuetRRFPlugin
2020-08-10 15:27:32,061 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrinterSettingsPlugin
2020-08-10 15:27:32,067 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OrientationPlugin
2020-08-10 15:27:32,070 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [374]: Loading all plugins took 0.6725301742553711 seconds
2020-08-10 15:27:32,077 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [160]: Looking for old configuration files to upgrade.
2020-08-10 15:27:32,171 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,174 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,182 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,216 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,220 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,224 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,228 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,231 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,234 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,238 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,241 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,244 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,247 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,251 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,254 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ./.DS_Store, skip it
2020-08-10 15:27:32,406 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ../custom_material.xml.fdm_material, skip it
2020-08-10 15:27:32,421 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [328]: Failed to get file version: ../generic_pla_175+%232.xml.fdm_material, skip it
2020-08-10 15:27:35,673 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [360]: Loading metadata into container registry took 2.910377025604248 seconds
2020-08-10 15:27:35,897 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [111]: Select build plate: 0
2020-08-10 15:27:35,900 - INFO - [MainThread] cura.CuraApplication.run [751]: Initializing machine error checker
2020-08-10 15:27:36,086 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [72]: Default machine actions added for machine definition [custom]
[... cut ...]
2020-08-10 15:35:24,608 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onReadyToPrint [289]: Охвофток-print | Ready to print
2020-08-10 15:35:24,608 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onReadyToPrint [292]: Охвофток-print | Sending gcode:M32 "0:/gcodes/b44560d5-b0a2-4a72-b6a3-52236024c0e0.gcode"
2020-08-10 15:35:24,630 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onPrintStarted [311]: Охвофток-print | Print started
2020-08-10 15:35:24,631 - ERROR - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._cleanupRequest [439]: _cleanupRequest called
```

zlowred, 2020-08-10:
Additional info: It doesn't look that the crash is STL-dependent, i.e. I see it with basically any model I try to print. Also, it looks like Cura won't crash if I use "Upload to ..." instead of "Print on ..."

zlowred, 2020-08-10:
And more info: when using "Upload to ..." I still see the last `ERROR` line in the log, but Cura doesn't crash:
```
2020-08-10 16:19:14,982 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [WARNING] Warning: Couldn't find previous extruder plan so as to set the standby temperature. Inserting temp command in earliest available layer.
2020-08-10 16:19:14,986 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [WARNING] Empty extruder plan detected! Discarding extrusion temperature command.
2020-08-10 16:19:16,090 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2020-08-10 16:19:16,167 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [678]: Slicing took 5.667165040969849 seconds
2020-08-10 16:19:16,168 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [679]: Number of models per buildplate: {0: 5}
2020-08-10 16:19:16,168 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [693]: See if there is more to slice...
2020-08-10 16:19:16,393 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Closing
2020-08-10 16:19:16,396 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [216]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2020-08-10 16:19:16,398 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [194]: Socket state changed to Closed
2020-08-10 16:19:16,398 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [844]: Backend quit with return code 0. Resetting process and socket.
2020-08-10 16:19:17,672 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-08-10 16:19:19,409 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Охвофток-upload | Filename set to: 22130f91-58aa-4204-9ff0-5cc5bf59ef07.gcode
2020-08-10 16:19:20,026 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Охвофток-upload | Loading gcode...
2020-08-10 16:19:20,341 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Охвофток-upload | Connecting...
2020-08-10 16:19:20,391 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [346]: Request [5b337ae47c6b44af827fcc0ec5c422e4] started
2020-08-10 16:19:20,395 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-08-10 16:19:20,424 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Охвофток-upload | Uploading...
2020-08-10 16:19:21,037 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [408]: Request [5b337ae47c6b44af827fcc0ec5c422e4] finished.
2020-08-10 16:19:21,040 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-08-10 16:19:21,040 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
2020-08-10 16:19:28,173 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadDone [245]: Охвофток-upload | Upload done
2020-08-10 16:19:28,176 - ERROR - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._cleanupRequest [439]: _cleanupRequest called
2020-08-10 16:21:20,166 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Охвофток-upload | Filename set to: 22130f91-58aa-4204-9ff0-5cc5bf59ef07.gcode
2020-08-10 16:21:20,784 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Охвофток-upload | Loading gcode...
2020-08-10 16:21:21,015 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Охвофток-upload | Connecting...
2020-08-10 16:21:21,056 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [346]: Request [578b1ded589645c1bbf4fd9c59f8117b] started
2020-08-10 16:21:21,059 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-08-10 16:21:21,060 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Охвофток-upload | Uploading...
2020-08-10 16:21:21,608 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [408]: Request [578b1ded589645c1bbf4fd9c59f8117b] finished.
2020-08-10 16:21:21,611 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-08-10 16:21:21,611 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
2020-08-10 16:21:29,667 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadDone [245]: Охвофток-upload | Upload done
2020-08-10 16:21:29,670 - ERROR - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._cleanupRequest [439]: _cleanupRequest called
```

zlowred, 2020-08-10:
Crash report from Mac OS:
```
Process:               cura [56708]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.6 (4.6.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [56708]
User ID:               414628

Date/Time:             2020-08-10 15:35:24.690 +0800
OS Version:            Mac OS X 10.15.5 (19F101)
Report Version:        12
Bridge OS Version:     4.5 (17P5300)
Anonymous UUID:        8956821C-6038-4DD8-B04D-930909B05430

Sleep/Wake UUID:       1EEFE8D2-2324-41A3-83F3-10DE1B11850E

Time Awake Since Boot: 3300000 seconds
Time Since Wake:       63000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00007fff35055177
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x7fff35055177:
    __TEXT                 00007fff34f47000-00007fff35000000 [  740K] r-x/r-x SM=COW  /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
--> __TEXT                 00007fff35000000-00007fff35200000 [ 2048K] r-x/rwx SM=COW  /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
    Submap                 00007fff35200000-00007fff40000000 [174.0M] r--/rwx SM=PRV  process-only VM submap

Thread 0 Crashed:
0   libsystem_kernel.dylib        	0x00007fff6ca5333a __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff6c96a93e raise + 26
2   libsystem_platform.dylib      	0x00007fff6cb045fd _sigtramp + 29
3   ???                           	000000000000000000 0 + 0
4   libQt5Core.5.dylib            	0x000000010da968b9 QMetaObject::activate(QObject*, int, int, void**) + 3113
5   libQt5Network.5.dylib         	0x000000010d742cf0 0x10d704000 + 257264
6   libQt5Core.5.dylib            	0x000000010da8eb32 QObject::event(QEvent*) + 146
7   libQt5Widgets.5.dylib         	0x00000001105290ba QApplicationPrivate::notify_helper(QObject*, QEvent*) + 314
8   libQt5Widgets.5.dylib         	0x000000011052b6ed QApplication::notify(QObject*, QEvent*) + 5181
9   PyQt5.QtWidgets.so            	0x0000000110002d62 sipQApplication::notify(QObject*, QEvent*) + 114
10  libQt5Core.5.dylib            	0x000000010da62ae4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
11  libQt5Core.5.dylib            	0x000000010da639be QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 846
12  libqcocoa.dylib               	0x000000131d94489e 0x131d917000 + 186526
13  libqcocoa.dylib               	0x000000131d945151 0x131d917000 + 188753
14  com.apple.CoreFoundation      	0x00007fff32910de2 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
15  com.apple.CoreFoundation      	0x00007fff32910d81 __CFRunLoopDoSource0 + 103
16  com.apple.CoreFoundation      	0x00007fff32910b9b __CFRunLoopDoSources0 + 209
17  com.apple.CoreFoundation      	0x00007fff3290f8ca __CFRunLoopRun + 927
18  com.apple.CoreFoundation      	0x00007fff3290eece CFRunLoopRunSpecific + 462
19  com.apple.HIToolbox           	0x00007fff3153dabd RunCurrentEventLoopInMode + 292
20  com.apple.HIToolbox           	0x00007fff3153d7d5 ReceiveNextEventCommon + 584
21  com.apple.HIToolbox           	0x00007fff3153d579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
22  com.apple.AppKit              	0x00007fff2fb85829 _DPSNextEvent + 883
23  com.apple.AppKit              	0x00007fff2fb84070 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
24  com.apple.AppKit              	0x00007fff2fb75d7e -[NSApplication run] + 658
25  libqcocoa.dylib               	0x000000131d944001 0x131d917000 + 184321
26  libQt5Core.5.dylib            	0x000000010da5e2f1 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 417
27  libQt5Core.5.dylib            	0x000000010da631b6 QCoreApplication::exec() + 390
28  PyQt5.QtWidgets.so            	0x0000000110003ce0 meth_QApplication_exec_(_object*, _object*) + 80
29  nl.ultimaker.cura             	0x000000010d1c6535 PyCFunction_Call + 165
30  nl.ultimaker.cura             	0x000000010d24560b PyEval_EvalFrameEx + 27963
31  nl.ultimaker.cura             	0x000000010d2494b2 0x10d171000 + 885938
32  nl.ultimaker.cura             	0x000000010d249d90 0x10d171000 + 888208
33  nl.ultimaker.cura             	0x000000010d244733 PyEval_EvalFrameEx + 24163
[... 1832 lines cut ...]
    0x7fff5bb46000 -     0x7fff5bb51ff2  com.apple.PerformanceAnalysis (1.243.2 - 243.2) <941698D6-EF00-3D59-8560-F160BC04B412> /System/Library/PrivateFrameworks/PerformanceAnalysis.framework/Versions/A/PerformanceAnalysis
    0x7fff5bb52000 -     0x7fff5bb7affb  com.apple.persistentconnection (1.0 - 1.0) <FFC20BB6-9CF3-39F2-AE5C-8DEE348FE0C1> /System/Library/PrivateFrameworks/PersistentConnection.framework/Versions/A/PersistentConnection
    0x7fff5bb7b000 -     0x7fff5bb89ffe  com.apple.PersonaKit (1.0 - 1) <C2FEE024-D4D9-3E2F-89A9-3F054E7A28BA> /System/Library/PrivateFrameworks/PersonaKit.framework/Versions/A/PersonaKit
    0x7fff5bd33000 -     0x7fff5bd33ffa  com.apple.PhoneNumbers (1.0 - 1) <08BB706C-0D00-3519-B038-0770DE877865> /System/Library/PrivateFrameworks/PhoneNumbers.framework/Versions/A/PhoneNumbers
    0x7fff5d5f3000 -     0x7fff5d623ff7  com.apple.pluginkit.framework (1.0 - 1) <6A417FEF-CEDE-3EE0-93ED-42680A31C886> /System/Library/PrivateFrameworks/PlugInKit.framework/Versions/A/PlugInKit
[... cut ...]
    0x7fff6074f000 -     0x7fff6076dff2  com.apple.shortcut (2.16 - 106) <51323823-C5EB-3EE0-AA81-8D633A3F75BD> /System/Library/PrivateFrameworks/Shortcut.framework/Versions/A/Shortcut
    0x7fff6076e000 -     0x7fff6078eff5  com.apple.sidecar-core (1.0 - 209.40.4) <A5BEBCF6-0B2F-3BA7-9F4C-A65E4AECB2E6> /System/Library/PrivateFrameworks/SidecarCore.framework/Versions/A/SidecarCore
    0x7fff6078f000 -     0x7fff607a1ff0  com.apple.sidecar-ui (1.0 - 209.40.4) <55486B3A-B65A-3023-89C7-A154A39D0D7E> /System/Library/PrivateFrameworks/SidecarUI.framework/Versions/A/SidecarUI
    0x7fff61

Kriechi, 2020-08-10:
@zlowred thanks for providing this information!
The last error line with `_cleanupRequest called` is not an error, just a debug message (I chose the wrong log level).

Unfortunately I cannot reproduce the crash. I can print/upload/simulate with my printer without problems.

I'm surprised about this hard crash of Cura (or actually pyQt5 / Qt5?). @fieldOfView @Ghostkeeper any ideas where and how to start debugging this? I would have expected a Python stacktrace - but this issues seems to be on a different level...

Ghostkeeper, 2020-08-11:
It's indeed actually Qt that crashes, seemingly while processing a signal event. However it does also indicate that it's crashing through a [SIGBUS](https://en.wikipedia.org/wiki/SIGBUS). And more specifically, it mentions a `KERN_PROTECTION_FAILURE`, which seems to be MacOS-specific lingo for the operating system being requested to access an inaccessible part of kernel memory. While `SIGBUS` is normally a software bug (meaning that it would be a problem in Qt), when I search for that `KERN_PROTECTION_FAILURE` I find only support threads that suggest it's a hardware failure, which would explain why Kriechi isn't able to reproduce the problem.

I found [this thread](https://www.mac-forums.com/threads/kern-protection-failure.242396/) suggesting that it could be a problem of the GPU overheating, which is plausible with Europe's weather at the moment. Could also be a driver issue, since those run in MacOS' equivalent of kernel mode. I also found lots of threads suggesting to run a hard disk scan for bad sectors, but that seems less likely to me considering that it's not crashing in code that handles anywhere nearby where it's reading from or writing to the hard disk.

zlowred, 2020-08-11:
@brianmichalk1 @broncosis could you by any chance share your RRF plugin configuration from Cura? Either a screenshot or just all configuration parameters (no IP address is needed if you're concerned about that so you can mask it). Trying to confirm some hypothesis for the crash.

zlowred, 2020-08-11:
@Ghostkeeper I suspect it could be some Unicode mis-handling somewhere in between of the plugin, Cura plugin engine, Cura UI and Qt. I can't see crashes anymore after I renamed my printer to only have basic English characters. Could be somewhere on the path mentioned above Unicode is not handled properly and characters wider than 1 byte causing memory corruption.

I don't believe it's a hardware issue, I consistently see the crash on 3 pieces of completely unrelated hardware (2017 15" Macbook Pro, 2019 16" MacBook Pro and latest iMac Pro desktop)

brianmichalk1, 2020-08-11:
I'll send everything I have.  Just tell me which directories you want.

zlowred, 2020-08-11:
@brianmichalk1 depending on your OS, cura.log from these directories:
Windows: C:\Users\<username>\AppData\Roaming\cura\4.6\cura.log
macOS: ~/Library/Application Support/Cura/4.6/cura.log
Linux: /home/<username>/.local/share/cura/4.6/cura.log

Plus, the screenshot of the configuration dialog where you configured your printer for the plugin.
[image]

Kriechi, 2020-08-11:
Yes, indeed it seems to be Unicode related!
I used the printer name `Охвофток` from @zlowred and was able to reproduce the crash on the first try!

broncosis, 2020-08-12:
Its simple 

http://10.0.1.27

And a simple 1 word Password 
With a number at the end

⁣Get BlueMail for Android ​

broncosis, 2020-08-12:
just to be safe I took the space out of the name and it still crashed

broncosis, 2020-08-12:
I am on a amd based desktop running Linux mint 19.3 

and the crashes happen 100% of the time with 4.6.2 installed via snap

And about 50% of the time with a 4.6.1 app image 

⁣Get BlueMail for Android ​

zlowred, 2020-08-12:
> just to be safe I took the space out of the name and it still crashed
> also the linux log file loctaion is not the same in cura installed with snap
> /home/~user~/snap/cura-slicer/13/.local/share/cura.log
> 
> [cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/5060081/cura.log)

These logs looks like a different crash, Duet seems to close the connection. Can you double-check your printer's IP address (in case it changed for whatever reason) and auth settings?

broncosis, 2020-08-12:
[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/5062825/cura.log)

here is a new copy log  this is from 4.6.2 so was the last it crashes 100% of the time
the problem is that I had changed the password by 2 characters for the screw shot 
and then made 3 atempts to slice and up load with different printer names before I caught on 
I then made several more after correcting

if there something else I can do to help

commit e5180432 (2020-08-29): fix Cura crashes with non-latin Message boxes

fixes #27

The `UM.Message` box seems to cause crashes with Unicode strings when showing two messages after another while
not properly removing the first one.
closed 2020-08-29 by Kriechi