Kriechi/Cura-DuetRRFPlugin #28: Print command adds extra quoations Duet v0.8.5

reneurossance, 2020-09-28:
We have a Fusion3 F400-S which is running on a Duet v.0.8.5 board.  I am able to connect to the the machine via Cura and DuetRRFPlugin and upload files.  However, if I attempt to print the file I get an error that it is unable to find the file.  I can see the file in "0:/gcodes/" from within Duet Web Control and I can print it from there successfully.  The only potential issue I can identify is that the missing file name in the error is double quoted.

- Error: GCode File ""0:/gcodes/test.gcode"" not found

I am wondering if extra quotes are being added preventing it from successfully printing directly from within Cura via DuetRRFPlugin.

**Expected behavior**
When pressing the "Print on Fusion3 F400-S" button the file should be uploaded and the print started.

**Version Information**
 - Operating System: macOS
 - Cura Version: 4.7.1
 - Cura-DuetRRFPlugin Version and Installation: 1.0.11
 - Duet RepRapFirmware Version: [e.g. 3.1.1, check with `M122`]

** Cura Log**

Kriechi, 2020-09-28:
@reneurossance you conveniently left out the most important information: whats your RRF version running on the Duet0.8.5?
I suspect its one of the older pre-1.20 series? I haven't tested those in a long time...

There is the following entry in the RRF 1.20 changelog:
> String parameters (e.g. filenames) can optionally be enclosed in double quote characters, in all GCode commands for which double quote characters are not compulsory

What happens if you open Duet Web Control and type this into the Console command prompt:
`M32 "0:/gcodes/test.gcode"`

reneurossance, 2020-09-28:
My bad.  Still new to this.  I've denoted the RRF version below.
- RRF Version:  FusionWare 1.17e R2c+8 (2017-02-14)

When I run the M32 command I get the following error:

`M32 "0:/GCODES/TEST.GCODE"`
`GCode file ""0:/GCODES/TEST.GCODE"" not found
Failed to open file "0:/GCODES/TEST.GCODE"`

reneurossance, 2020-09-28:
The following works without quotes:
`M32 0:/GCODES/TEST_CUBE.GCODE`
`File 0:/GCODES/TEST_CUBE.GCODE selected for printing`

Kriechi, 2020-09-29:
@reneurossance thanks for providing the info!
Unfortunately there are a few issues: it looks like your printer came with a custom RepRapFirmware (not the official build), and even worse: it's from 2017. I don't think I can or want to support such old firmwares...

I could try implementing a fix for the quotes, but I'm certain there will be other incompatibilities along the line which would result in a big mess...

Does your printer vendor provide a new firmware that you could upgrade to?
Do you know which custom changes FusionWare implemented, maybe you can upgrade to the official RRF versions?

reneurossance, 2020-09-29:
Thank you for your time, efforts, and thoughts.  I am reaching out to Fusion3 (https://www.fusion3design.com) to see what they might be able to do or if they have any updated firmware but in the past their statement has been that they only support Simply3D and are unable to assist further, so, we'll see.

closed 2020-09-29 by Kriechi