Kriechi/Cura-DuetRRFPlugin #29: No Connect to dialog button...??

jaymcd, 2020-11-25:
I installed the plugin from the Marketplace and tried to follow all the steps on your github page but I don't have a connect to dialog box other than the Octoprint one. I have a new printer with a Duet 2 Maestro running V3 firmware and using Cura 4.8.0.

Is the Octoprint plugin interfering or ?

Kriechi, 2020-11-25:
I'm in the middle of upgrading the plugin for Cura 4.8 - the README you are seeing is already containing the new information.
Please refer to this README version that is currently valid for the plugin version in the Cura Marketplace as of 2020-11-25.

jaymcd, 2020-11-25:
Nice! Works !!!

Jay M McDonald
jaymcd@shaw.ca

https://github.com/Kriechi/Cura-DuetRRFPlugin/blob/f5c6437e9d5a6ff07228c40403d9777c12152d8e/README.md

Kriechi, 2021-01-19:
v1.2.0 is released.

closed 2021-01-19 by Kriechi