Kriechi/Cura-DuetRRFPlugin #30: Strange user experience compared to previous versions

brianmichalk1, 2020-11-30:
I have read that Kriechi is migrating, so this may be a moot point.
I downloaded the latest tar from Git on 11/30/2020.
I have three printers, all Duet.  The workflow now requires that I activate one printer, and that's the only one that shows up in the right lower corner.  If a printer is activated that does not have a connection in the plugin, it is not listed.
In the past, any Duet printer would show up for printing.  Should I sit and wait, or is this the expected workflow moving forward?

[image]

Kriechi, 2020-11-30:
Thats the intended migration.
The Cura-DuetRRF plugins now uses the Cura printers - you can select the Cura printer in the top-left.
If you select a printer that has the plugin configured, you will get the correct "Print" blue buttons.
If you select a printer that has not been configured, you will only get the default buttons to "Save to ...".
Is this working for you, are you expecting different behaviour?

brianmichalk1, 2020-11-30:
Hmm.  If I set the server address to http://rmaxv4, then go to artemis, and configure it, if I go back to RMaxV4, it has http://artemis as its server address..

[image]

Kriechi, 2020-11-30:
That looks like a bug.

Kriechi, 2021-01-04:
This should be fixed on git-master already - I will publish a new version to the Cura Marketplace in the next days or week.

closed 2021-01-04 by Kriechi