Kriechi/Cura-DuetRRFPlugin #31: Multiple printer support

zylantha, 2021-01-04:
While it's possible to add two (or more) different printers to Cura-DuetRRFPlugin, there's no way to map each of those printers to their different Cura printer profiles.

This means that even after switching Cura printers, I need to remember to change the current active output over to the correct printer (as it will remain as the previous printer).  

It would be great if when you configure the RepRapFirmware printers within the plugin, you could also map those to the associated Cura printer, so that when you click on "Print" to print via the plugin, it will automatically select and send it to the correct printer.

Without this automatic switching, I am constantly accidentally sending print jobs configured for one printer over to the other printer (which in my case fails as they are incompatible setups).

Kriechi, 2021-01-04:
The current git-master already has this new mapping - and I'm about to finish a few smaller changes to the settings storage.
Duplicate of #8

closed 2021-01-04 by Kriechi

Kriechi, 2021-01-09:
@zylantha I just pushed the latest changes to the "next" branch: https://github.com/Kriechi/Cura-DuetRRFPlugin/tree/next
Do you have a few minutes to test the plugin and report if this solves your configuration setup with multiple printers?

zylantha, 2021-01-14:
Yes I have a go

zylantha, 2021-01-14:
First test is positive and is working as expected.  I am tag teaming print jobs between my two printers so will continue testing over the next week and report any issues.