Kriechi/Cura-DuetRRFPlugin #32: Cura crashes on startup on DuetRRFPlugin version 1.1.0

konskarm, 2021-01-11:
**Describe the bug**
Cura crashes immediately on startup

**To Reproduce**
Steps to reproduce the behavior:
1. Load the plugin (version 1.1.0) in `%appdata%/cura/4.8/plugins`
2. Start Cura

**Expected behavior**
Cura doesn't crash

**Actual behavior**

[image]

**Version Information**
 - Operating System: **Windows 10**
 - Cura Version: **4.7.1 / 4.8**
 - Cura-DuetRRFPlugin Version and Installation: **1.1.0**
 - Duet RepRapFirmware Version: **N/A**

**Cura Log**

```
  File "C:\Users\<user>\AppData\Roaming\cura\master\plugins\DuetRRFPlugin\DuetRRFPlugin\DuetRRFPlugin.py", line 129, in _checkDuetRRFOutputDevices
    s = get_config()
  File "C:\Users\<user>\AppData\Roaming\cura\master\plugins\DuetRRFPlugin\DuetRRFPlugin\DuetRRFSettings.py", line 23, in get_config
    s, printer_id = _load_prefs()
  File "C:\Users\<user>\AppData\Roaming\cura\master\plugins\DuetRRFPlugin\DuetRRFPlugin\DuetRRFSettings.py", line 14, in _load_prefs
    s = json.loads(p.getValue(DUETRRF_SETTINGS))
  File "C:\Python\Python37\lib\json\__init__.py", line 341, in loads
    raise TypeError(f'the JSON object must be str, bytes or bytearray, '
TypeError: the JSON object must be str, bytes or bytearray, not NoneType
```

Kriechi, 2021-01-11:
Thanks for logging this bug!
I believe this is actually v1.2.0 from the `next` branch.
I just pushed a fix & rebased that branch.

closed 2021-01-11 by Kriechi

Kriechi, 2021-01-19:
v1.2.0 is released.