Kriechi/Cura-DuetRRFPlugin #34: Plugin Update continues to prompt for unmapped settings after setting up printer.

antbarney, 2021-01-21:
Popup shows every time I open Cura even after Deleting unmapped settings and reconfiguring printer.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Clear Settings as popup suggests, and restart Cura
2. Open Cura Printer Settings and Activate printer 
3. Click on Connect Duet RepRapFirmware button, add Duet Address and click Save.
[image]
3. After clicking Save plugin still will not allow upload to printer.
4. Restart Cura and popup returns.
[image]
5. But now the plug does work and I can upload to the Duet via the Plugin.

**Expected behavior**
Be able to properly configure the Plugin so that it does not think there are unmapped settings.
Only thing I noticed is that I do not have a Save & Test button mine only shows "Save"

**Version Information**
 - Operating System: Windows 10
 - Cura Version: 4.8
 - Cura-DuetRRFPlugin Version and Installation: 1.2.0 Installed via Market Place
 - Duet RepRapFirmware Version: RepRapFirmware for Duet 2 WiFi/Ethernet version 3.2 running on Duet WiFi 1.02 or later

[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/5847581/cura.log)

Based on your Cura version, please drag&drop the log file into the issue, or copy&paste all lines with `DuetRRFPlugin`:
Windows: C:\Users\<username>\AppData\Roaming\cura\<latest version>\cura.log
macOS: ~/Library/Application Support/Cura/<latest version>/cura.log
Linux: /home/<username>/.local/share/cura/<latest version>/cura.log

Kriechi, 2021-01-21:
Thanks reporting this.
This seems to be a race condition while Cura is still starting up and loading all the printers in the background.
Currently I have a 2second timer before the plugin checks for unmapped settings.

I've pushed a fix and will release a new version soon.

Kriechi, 2021-01-26:
v1.2.1 has landed!

closed 2021-01-26 by Kriechi

mydevpeeps, 2021-01-28:
I am still experiencing this after installing 1.2.1.

Kriechi, 2021-01-28:
@mydevpeeps how soon after starting Cura do you see the popup message appear?
v1.2.1 should have made this hook into the Cura event `initializationFinished` and then (because I already am extra paranoid) wait another 10 seconds before checking the settings and displaying the popup...
Please upload your cura log file.

mydevpeeps, 2021-01-28:
** Summary **
it happens about 6 seconds after the ui opened, and about 2 second after the profile loaded.

** Reproduction **
it shows in the marketplace I have the latest 1.2.2 installed:

[image]

I get this message pop up (password removed from screenshot):

[image]

If I try to go to extensions -> DuetRRF like it instructs me, to choose "Delete unmapped settings", that option does not exist:

[image]

**Version?**
One odd thing about both the logs and the plugin json file, is they both show this as version 1.2.0 even though I had updated to 1.2.1. It's as if both versions are loaded at once?

** Upgrade to 1.2.2 **
```2021-01-28 14:37:54,661 - DEBUG - [MainThread] Toolbox.src.Toolbox._createDialog [216]: Marketplace: Creating dialog [Toolbox.qml].
2021-01-28 14:37:55,086 - DEBUG - [MainThread] Toolbox.src.Toolbox._makeRequestByType [544]: Requesting [packages] metadata from server.
2021-01-28 14:37:55,092 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [124]: No auth data to retrieve the access_token from
2021-01-28 14:37:55,095 - WARNING - [MainThread] UM.Logger.warning [128]: Cannot add authorization to Cloud Api request
2021-01-28 14:37:55,096 - DEBUG - [MainThread] Toolbox.src.Toolbox._makeRequestByType [544]: Requesting [authors] metadata from server.
2021-01-28 14:37:55,099 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [124]: No auth data to retrieve the access_token from
2021-01-28 14:37:55,103 - WARNING - [MainThread] UM.Logger.warning [128]: Cannot add authorization to Cloud Api request
2021-01-28 14:37:55,104 - DEBUG - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [270]: Found a plugin that was installed with the old plugin browser: VersionUpgrade47to48
2021-01-28 14:38:52,080 - DEBUG - [MainThread] Toolbox.src.Toolbox._update [429]: Updating package [DuetRRFPlugin]...
2021-01-28 14:38:52,081 - INFO - [MainThread] Toolbox.src.Toolbox.startDownload [556]: Attempting to download & install package from https://api.ultimaker.com/cura-packages/v1/cura/v7.4.0/packages/DuetRRFPlugin/download.
2021-01-28 14:38:52,086 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [124]: No auth data to retrieve the access_token from
2021-01-28 14:38:52,090 - WARNING - [MainThread] UM.Logger.warning [128]: Cannot add authorization to Cloud Api request
2021-01-28 14:38:53,150 - DEBUG - [MainThread] Toolbox.src.Toolbox._onDownloadProgress [673]: new download progress 213075 / 213075 : 100.0%
2021-01-28 14:38:53,150 - DEBUG - [MainThread] Toolbox.src.Toolbox._onDownloadProgress [673]: new download progress 213075 / 213075 : 100.0%
2021-01-28 14:38:53,230 - INFO - [MainThread] Toolbox.src.Toolbox._onDownloadComplete [676]: Download complete.
2021-01-28 14:38:53,234 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [601]: Found potential package.json file '/package.json'
2021-01-28 14:38:53,239 - DEBUG - [MainThread] UM.PackageManager.getPackageLicense [629]: Found potential license file '/files/plugins/DuetRRFPlugin/LICENSE'
2021-01-28 14:38:55,633 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [601]: Found potential package.json file '/package.json'
2021-01-28 14:38:55,638 - INFO - [MainThread] UM.PackageManager.installPackage [447]: Package [DuetRRFPlugin] version [1.2.2] is scheduled to be installed.
2021-01-28 14:38:55,656 - INFO - [MainThread] UM.PackageManager._saveManagementData [259]: Package management file C:\Users\localadmin\AppData\Roaming\cura\4.8\packages.json was saved
2021-01-28 14:38:55,673 - DEBUG - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [270]: Found a plugin that was installed with the old plugin browser: VersionUpgrade47to48
2021-01-28 14:38:57,430 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura```

** Startup **
2021-01-28 14:39:08,796 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.8.0

_snipped_
```2021-01-28 14:39:11,038 - DEBUG - [MainThread] DuetRRFPlugin.register [18]: DuetRRFPlugin version: 1.2.0
2021-01-28 14:39:11,045 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin DuetRRFPlugin
2021-01-28 14:39:15,241 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._checkDuetRRFOutputDevices [140]: DuetRRF is active for printer: id:Prusa i3 Factory, name:HICTOP Prusa i3 MK1
2021-01-28 14:39:15,242 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.__init__ [109]: New duetrrf-print DuetRRFOutputDevice created | URL: http://192.168.4.22/ | Duet password: set | HTTP Basic Auth: user:<empty>, password:<empty>
2021-01-28 14:39:15,242 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._resetState [481]: called
2021-01-28 14:39:15,243 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.__init__ [109]: New duetrrf-simulate DuetRRFOutputDevice created | URL: http://192.168.4.22/ | Duet password: set | HTTP Basic Auth: user:<empty>, password:<empty>
2021-01-28 14:39:15,243 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._resetState [481]: called
2021-01-28 14:39:15,243 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.__init__ [109]: New duetrrf-upload DuetRRFOutputDevice created | URL: http://192.168.4.22/ | Duet password: set | HTTP Basic Auth: user:<empty>, password:<empty>
2021-01-28 14:39:15,244 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._resetState [481]: called
2021-01-28 14:39:28,977 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._check_unmapped_settings [47]: called
2021-01-28 14:39:28,978 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._check_unmapped_settings [61]: Unmapped settings found!
2021-01-28 14:39:28,978 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._showUnmappedSettingsMessage [69]: called: dict_keys(['Prusa i3'])
2021-01-28 14:41:34,456 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._showUnmappedSettingsMessage [69]: called: dict_keys(['Prusa i3'])
```

Kriechi, 2021-01-28:
Thanks for the good amount of information!

**About the version number:**
Thanks for pointing that out - looks like I forgot to bump the number in one of the two places that need it. Should have no negative effect though. I already bumped it for the next time.

**About the log:**
Please check if you also find the timestamp / log line that says `Booting Cura took`

Your log clearly shows that it found and recognized a printer `DuetRRF is active for printer: id:Prusa i3 Factory, name:HICTOP Prusa i3 MK1`. Do you have more than one printer names "Prusa i3" (maybe also if you renamed it, but the original name was Prusa?)

I guess the second log entry `_showUnmappedSettingsMessage` was you clicking on the menu item?

Could you please provide a small piece of your `cura.cfg` config file - it is in the same place as the log file:
```
[duetrrf]
instances = <a large JSON string>
```
Make sure to remove/replace your passwords!

mydevpeeps, 2021-01-28:
** logs **
2021-01-28 14:39:19,328 - DEBUG - [MainThread] cura.CuraApplication.run [833]: Booting Cura took 12.814523220062256 seconds

** printers **
[image]

** config **
[output_devices]
last_used_device = duetrrf-configure

[duetrrf]
instances = {"Prusa i3 Factory": {"url": "http://192.168.4.22/", "http_password": "", "http_user": "", "duet_password": "redacted"}, "Prusa i3": {"url": "http://192.168.4.22/", "http_password": "", "http_user": "", "duet_password": "reprap"}}

mydevpeeps, 2021-01-28:
I am wondering if this happened because at some point in the past, I added the printer as Prusa I3, and then I renamed it later after adding the plugin, and then I upgraded the plugin to 1.2.1.

Kriechi, 2021-01-28:
Yep - so the message complaining about unmapped settings is correct.
Did you click the `Delete unmapped settings` menu item already?

mydevpeeps, 2021-01-28:
It's not an option (see previous screenshot) and i honestly can't remember if I did that after the upgrade to 1.2.1. would the logs tell me?

Kriechi, 2021-01-28:
If you clicked the `Delete unmapped settings` menu item, the logs should have a line with `_deleteUnmappedSettings`.

Kriechi, 2021-01-28:
So there is apparently a bug that the menu item is not available. I was able to reproduce this issue.

mydevpeeps, 2021-01-28:
I don't have that, so I never clicked it. Thanks.

Kriechi, 2021-01-30:
Thanks @mydevpeeps for helping me track down this problem!
This should fix it: 4fccb8be1cc82a3e769b1d00ea91afbf8106e6d4
The underlying issue was that I wrongly assumed that Uranium (the UI framework behind Cura) can dynamically add menu bar items. It can not. Therefore the `Delete unmapped settings` menu item never showed. I now changed it to be a button on the message popup instead.

New release is submitted to Ultimaker for review - should go live in a few days.