Kriechi/Cura-DuetRRFPlugin #35: can't print unless selecting another printer and then back again

mydevpeeps, 2021-01-27:
I just printed this morning on the previous plugin version. I updated the plugin to the latest from the marketplace and restarted cura. I get the Connect Duet RepRapFirmware button, I click it, I put in my config. I click Save Config (note that I do not have any Save & Test as the instructions do, just a Save Config). I save Config. I slice the file, and then the button says "DuetRRF Plugin" instead of "Print on <printer name>".  If I change it to "Save Gcode" then in the dropdown I have "Configure Duet RepRapFirmware...". 

I select another printer, and then select this printer again, and now all of a sudden the print on <printer name> shows up... 

[image]

Kriechi, 2021-01-27:
Please provide the information we ask for in the bug report template:
https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/new?assignees=&labels=&template=bug_report.md&title=

mydevpeeps, 2021-01-27:
I hit edit and it saved it as a new issue. No idea why. Please close this one as a dupe.

closed 2021-01-27 by Kriechi