Kriechi/Cura-DuetRRFPlugin #37: DuetRRF plugin not available in Marketplace(4.9.1/4.10)

wolfgangmauer, 2021-07-03:
The great!!! Duet RRF plugin is not available in the Marketplace(4.9.1/4.10) :-(

Kriechi, 2021-07-03:
Does this work for you?
https://marketplace.ultimaker.com/app/cura/plugins/thomas_kriechbaumer/DuetRRFPlugin

Kriechi, 2021-07-03:
Also shows up just fine in Cura 4.10.0 on macOS, see this screenshot I just took with an empty/fresh/clean Cura profile:
[image]

wolfgangmauer, 2021-07-03:
Sorry, it shows up in the Marketplace, but with 4.8 it looks like this
[image]
with 4.9.1/4.10 the menu is not availible
[image]
And after slicing i only get
[image]

Kriechi, 2021-07-03:
1. The plugin settings moved to the "Manage Printer" section a while ago
2. The menu item should still be there - and it should tell you about the move.
3. You might have to reconfigure your printer settings if you clicked-away or deleted the old settings without reading the upgrade instructions.

Please check in the Marketplace again under "installed" plugins - maybe it is disabled?

wolfgangmauer, 2021-07-03:
Sorry for "spam", the plugin was disabled

closed 2021-07-03 by wolfgangmauer