Kriechi/Cura-DuetRRFPlugin #39: Mehrere Drucker auswählen mit den Plugin

steirerman, 2021-12-11:
Halllo

Ich möchte mehrere Drucker über das Plugin auswählen um jeden Drucker einen Job zuweisen zu können, ist das möglich ?
I would like to select several printers via the plugin in order to be able to assign a job to each printer, is that possible?

Kriechi, 2021-12-12:
Could you elaborate or be more specific please?
You can already configure multiple physical printers with Cura, and add RRF configuration to each one of them. Then simply slice your model, and use the button options to send this job to the selected printer.

steirerman, 2021-12-13:
Tut mir leid, das war ein Gedankenfehler von mir, wollte einen Job slicen und dann über das RRF plugin an mehrere Drucker verteilen. Ist ja technisch nicht möchlich, da beim RRF plugin keine Druckerkonfigurationen hinterlegt sind.
I'm sorry, that was a mistake of my thinking, wanted to slice a job and then distribute it to several printers via the RRF plugin.  This is technically not possible, as there are no printer configurations stored in the RRF plugin.

closed 2021-12-13 by steirerman