Kriechi/Cura-DuetRRFPlugin #40: Cannot upload to printer with DuetRRF

NikAWing, 2022-01-23:
I already opened a bug report on Cura git, but I was directed to write it here, so:

Hey all,

I'm kinda new to reporting bugs or occuring errors, so I hope I'm doing it right.

This problem started for me at least 3 Cura releases ago, currently I use 4.13.0 on Win7 x64.
I have a printer with Duet3 controller and RRF.
CuraRRF version is from marketplace 1.0.11 and RRF is the newest 3.4.0beta7, I can not tell if the problem occurs because of an RRF update tbh.
Of course, I use the DuetRRF plugin to send g-code files to the printer.
This worked before but suddenty it makes Cura crash (Windows shows Qt5network.dll in the crash information).

so it's Slice -> Upload to printer button -> filename ok -> crash

I can access the printer via webbrowser and also upload files using the web gui fine.

I tried it with disabled firewall etc without any effect.

With best regards,
Nik

at the end of stderr.log I see this:

> 2022-01-22 14:33:06,486 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [764]: Slicing took 1.2980740070343018 seconds
    2022-01-22 14:33:06,486 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [765]: Number of models per buildplate: {0: 1}
    2022-01-22 14:33:06,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [779]: See if there is more to slice...
    2022-01-22 14:33:06,490 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
    2022-01-22 14:33:06,493 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
    2022-01-22 14:33:06,497 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
    2022-01-22 14:33:06,509 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend finished slicing. Resetting process and socket.
    2022-01-22 14:33:07,255 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.020000934600830078 seconds
    2022-01-22 14:33:10,080 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Nik Duet 3-upload | Filename set to: Nik Duet3 - ABS_Boden.gcode
    2022-01-22 14:33:10,081 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [118]: 'info/send_slice_info' is turned off.
    2022-01-22 14:33:10,081 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [186]: ArcWelder is not enabled
    2022-01-22 14:33:10,082 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Nik Duet 3-upload | Loading gcode...
    2022-01-22 14:33:10,281 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Nik Duet 3-upload | Connecting...
    2022-01-22 14:33:10,334 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [202]: Nik Duet 3-upload | rr_connect failed with errorCode 203
    2022-01-22 14:33:10,335 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [204]: Nik Duet 3-upload | errorCode indicates Duet3+SBC - let's try the DuetSoftwareFramework API instead...
    2022-01-22 14:33:10,504 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Nik Duet 3-upload | Uploading...
    Windows fatal exception: access violation

> Thread 0x00002fb8 (most recent call first):
    File "C:\Program Files\Ultimaker Cura 4.13.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
    File "threading.py", line 870 in run
    File "threading.py", line 932 in _bootstrap_inner
    File "threading.py", line 890 in _bootstrap

> Thread 0x00003998 (most recent call first):
    File "threading.py", line 306 in wait
    File "zeroconf_init_.py", line 2534 in wait
    File "zeroconf_init_.py", line 1715 in run
    File "threading.py", line 932 in _bootstrap_inner
    File "threading.py", line 890 in _bootstrap

> Thread 0x000025c0 (most recent call first):
    File "threading.py", line 306 in wait
    File "threading.py", line 558 in wait
    File "C:\Program Files\Ultimaker Cura 4.13.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
    File "threading.py", line 870 in run
    File "threading.py", line 932 in _bootstrap_inner
    File "threading.py", line 890 in _bootstrap

> Thread 0x000006b4 (most recent call first):
    File "zeroconf_init_.py", line 1369 in run
    File "threading.py", line 932 in _bootstrap_inner
    File "threading.py", line 890 in _bootstrap

> Thread 0x000023f8 (most recent call first):
    File "C:\Program Files\Ultimaker Cura 4.13.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
    File "threading.py", line 870 in run
[... 35 lines cut ...]
> Current thread 0x000028d8 (most recent call first):
    File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 396 in exec_
    File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 877 in run
    File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 236 in
    File "Console.py", line 36 in run
    File "startup.py", line 74 in run

cura.log has this at the end:

> 2022-01-22 14:33:06,486 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [764]: Slicing took 1.2980740070343018 seconds
    2022-01-22 14:33:06,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [765]: Number of models per buildplate: {0: 1}
    2022-01-22 14:33:06,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [779]: See if there is more to slice...
    2022-01-22 14:33:06,490 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
    2022-01-22 14:33:06,493 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
    2022-01-22 14:33:06,497 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
    2022-01-22 14:33:06,509 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend finished slicing. Resetting process and socket.
    2022-01-22 14:33:07,256 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.020000934600830078 seconds
    2022-01-22 14:33:10,080 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Nik Duet 3-upload | Filename set to: Nik Duet3 - ABS_Boden.gcode
    2022-01-22 14:33:10,081 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [118]: 'info/send_slice_info' is turned off.
    2022-01-22 14:33:10,081 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [186]: ArcWelder is not enabled
    2022-01-22 14:33:10,082 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Nik Duet 3-upload | Loading gcode...
    2022-01-22 14:33:10,281 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Nik Duet 3-upload | Connecting...
    2022-01-22 14:33:10,334 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [202]: Nik Duet 3-upload | rr_connect failed with errorCode 203
    2022-01-22 14:33:10,335 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [204]: Nik Duet 3-upload | errorCode indicates Duet3+SBC - let's try the DuetSoftwareFramework API instead...
    2022-01-22 14:33:10,504 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Nik Duet 3-upload | Uploading...

Kriechi, 2022-01-23:
@NikAWing your plugin version is pretty outdated - the most recent one in the Marketplace is v1.2.4.
Please update the plugin to have the most recent matching versions of each.
I haven't tested yet with Cura 4.13, if you can, please also test with older versions of Cura.

NikAWing, 2022-01-23:
@Kriechi ok, now I have to ask this stupid question ;) 
How do I update it? Usually Cura marketplace notifies me about updates, DuetRRF was never listed, nor does it show an update button :o

Kriechi, 2022-01-24:
@NikAWing did you maybe install it manually? https://github.com/Kriechi/Cura-DuetRRFPlugin#manual-installation
You can try deleting the plugin from this folder, restartin Cura, and then installing from the Marketplace.

NikAWing, 2022-01-25:
@Kriechi no, always used the marketplace.
The changelog.md file in my plugin folder also just listed up to 1.0.11 so I did what you suggested.
I see there also was a major change with 1.2.0 concerning HttpRequestManager.
I'll try to test it later after work :)

NikAWing, 2022-01-25:
@Kriechi Updating fixed the problem! Now I need to find out why I did not see the update in marketplace. This morning, when I deleted the old plugin and re-installed it, I saw an update for another plugin, so whatever reason it was, I'll monitor any new DuetRRF updates and if they'll be listed in Cura :)

closed 2022-01-25 by NikAWing