Kriechi/Cura-DuetRRFPlugin #42: Cura 5.0 support

nallath, 2022-02-15:
Since I was reviewing your plugin, I also did a quick check if you were still using QtControls 1.0, since that will no longer be supported in the next Cura release (5.0) due to the upgrade to Qt6. 

I noticed that you still use controls 1.0, but luckily the interface is relatively simple. If you upgrade them to controls 2.0, it will still be backwards compatible.

Kriechi, 2022-02-15:
@nallath thanks for the heads-up! I briefly checked the upgrade path, and I got it mostly working, but the UI looks quite different... buttons are larger, textfields higher and with more padding, default button flag is missing, mouse-drag-to-select-text not working, text selection color not readable...

Do you have any pointers to upgrade docs or how did you handle this for Uranium+Cura?

Kriechi, 2022-02-15:
Old / Controls 1.0:
[image]

New / Controls 2.0:
[image]

nallath, 2022-02-16:
Yeah, there will be some visual changes, you really cant avoid that. The controls 2.0 no longer have system styling. 

We are currently working on making sure that there are re-usable components that will have the standard Cura style (see: https://github.com/Ultimaker/Cura/pull/10788), but those will obviously not be back ported for older versions.

As for customizing the textfield, you can find some more info about it here https://doc.qt.io/qt-5/qtquickcontrols2-customize.html#customizing-textfield

Kriechi, 2022-06-08:
fixed with the last release.