Kriechi/Cura-DuetRRFPlugin #44: Printer not showing to add

Monroe07, 2022-06-01:
Cura 5.0
Board: Duet 2 WiFi (2WiFi)
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.4.0 (2022-03-15)
Duet WiFi Server Version: 1.26 
Windows 11
Ultimaker 3 already added in Cura (Don't know if that detail is relevant)

When adding the printer in Cura, the printer does not show up in "Printer" section to "activate". No option in Cura discovers the printer.

[image]

Kriechi, 2022-06-02:
Can you post a screenshot?
You need to add a non-Ultimaker printer - usually using the "custom" category. If it is your only printer, you don't need to "activate" it. Simply select it in the list, and the config button should show up.

Monroe07, 2022-06-06:
I got it working. So to clarify, you have to select the "custom" section within the "Add a non-networked printer" section, which is counter intuitive when the purpose is to add a networked [3rd party] printer( I know, not your lane ). Should probably add a note in the documentation about doing that by either adding a printer using the main screen printer selection drop-down and choosing a custom "non-networked" option, or doing the same selection within the preferences.

Kriechi, 2022-06-08:
There is already this section in the README:
> **Do NOT try to add a new "networked printer"!** This is only for Ultimaker printers.

If you have any suggestion to improve this even further, I'm happy to merge a PR!