Kriechi/Cura-DuetRRFPlugin #48: Rename file

Benouz63, 2023-04-05:
Hello, first of all thank you for this great plugin, it is functional. on the other hand, for it to work for my machines, I have to add "file/" each time in front of the file name for the upload. it is very restrictive. is it possible to modify the code of the plugin so that it adds it automatically? Thanks in advance and sorry for the inconvenience

Kriechi, 2023-04-05:
Happy to hear you are enjoying the plugin! 🎉 

What kind of file prefix do you need? Why do you "have to add "file/" for each upload?

You can certainly edit the source code directly in your cura profile, but it most likely gets overwritten when I release the next update.

Benouz63, 2023-04-05:
the manufacturer of my printers has organized the operation with two folders for the gcodes, in the gcodes folder. I have "file" and "reprint". if i upload the file without prefix as it arrives in gcode root folder which i don't have access by machine screen. in the upload window of your plugin I have to add "file/filename.gcode" so that it appears on the screen. I wanted to automate the addition of "file/" before the file name in the upload window. is this understandable as an explanation? (I am French, thank you Google translation).

Kriechi, 2023-04-05:
Would this other plugin together with the Duet plugin work for you? https://github.com/fieldOfView/Cura-CustomJobPrefix

closed 2023-04-22 by Kriechi