Kriechi/Cura-DuetRRFPlugin #49: cura5.3.1

Olliftw, 2023-04-22:
is the duetRRFplugin not supported in 5.3.1? 
the plugin isnt showing eventouch already tryed reinstalling the pluggin etc.

grtz

Kriechi, 2023-04-22:
Yes, the plugin supports Cura 5.3.1.
It is available in the Marketplace:
[image]

Olliftw, 2023-04-22:
Wierd, cant get it to show up at the machine settings window, uninstalled the plugin, restarted cura and nothing, in marketplace it does show installed and up to date

Verstuurd vanaf mijn iPhone

[Screenshot 2023-04-22 at 16 32 46]<https://user-images.githubusercontent.com/114300/233790942-1ace63bb-dd5e-41ae-b0b1-ffa0461d329b.png>

Kriechi, 2023-04-22:
Please provide all the information from the bug report template: https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/new?template=bug_report.md

Olliftw, 2023-04-22:
[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/11301881/cura.log)

Kriechi, 2023-04-22:
Odd - I see that the plugin seems to be installed (and also uninstalled) in the log, but I don't see it being loaded.
I suspect that your Cura settings and profile might be corrupt or in an inconsistent state.

Please try using a clean Cura profile: https://support.makerbot.com/s/article/1667337567843

closed 2024-01-20 by Kriechi