Kriechi/Cura-DuetRRFPlugin #51: http auth crash on cura 5.4

Thijskunst, 2023-11-20:
plugin causes cura to crash when sending file to a printer with http basic authentication enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. load a 3d model
2. slice
3. click 'print on printername' or 'upload to printername'
4. See error

**Expected behavior**
print starts on printer or file gets uploaded to printer

**Version Information**
 - Operating System: windows 11 pro 22H2
 - Cura Version: 5.4.0
 - Cura-DuetRRF Plugin Version and Installation: 1.2.3 installed via Marketplace
 - Duet RepRapFirmware Version:  RepRapFirmware for STM32F4 based Boards 3.5beta2+_103 (2023-02-11)
 Duet WiFi Server Version: 1.27-02S32-D

**error traceback**
Traceback (most recent call last):
  File "C:\Users\thijs\AppData\Roaming\cura\5.4\plugins\DuetRRFPlugin\DuetRRFPlugin\DuetRRFOutputDevice.py", line 234, in _onFilenameAccepted
    self._send('rr_connect',
  File "C:\Users\thijs\AppData\Roaming\cura\5.4\plugins\DuetRRFPlugin\DuetRRFPlugin\DuetRRFOutputDevice.py", line 137, in _send
    headers['Authorization'] = 'Basic ' + base64.b64encode(auth)
TypeError: can only concatenate str (not "bytes") to str

**Cura Log**

[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/13419459/cura.log)

Kriechi, 2023-11-25:
Thanks for reporting this @Thijskunst!

In your local file `C:\Users\thijs\AppData\Roaming\cura\5.4\plugins\DuetRRFPlugin\DuetRRFPlugin\DuetRRFOutputDevice.py` n line 137, please change the line to:

```python
headers['Authorization'] = 'Basic ' + base64.b64encode(auth).decode()
```

Save the file and restart Cura.

Please try this bug fix - if it works for you, I'll commit a fix and release a new version.

Thijskunst, 2023-12-12:
thanks for your reply,

sorry for my slow reply. I was in the proces of changing the kinematics, so i wasnt able to print anything.
i tested it and the plugin works now without any issue.

commit aa0dc266 (2024-01-20): fix HTTP Basic Auth

closes #51
closed 2024-01-20 by Kriechi