Kriechi/Cura-DuetRRFPlugin #52: Tumbnail generated in G-code for all printers.

3D-Printguy, 2024-01-15:
**Describe the bug**
No matter what printer i Choose in Cura, a thumbnail is generated in the G-code output. Only one of my printers are running duet/RRF. The others are older 8-bit board marlin and repetier firmware. The old printers can not handle the long comment section in the start of the G-code and has to be manually removed.

I believe this started when I upgraded to Cura 5.6 but I'm not 100% sure.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a mix of reprap and marlin printers configured in Cura.
2. Choose a marlin printer
3. Slice and save G-code
4. Check G-code for thumbnail

**Expected behavior**
I would expect the plugin to only generate thumbnail for reprap printers. Or there could be an option to disable thumbnails.

**Version Information**
 - Operating System: Windows 11
 - Cura Version: 5.6.0
 - Cura-DuetRRF Plugin Version and Installation: 1.2.9 installed via Marketplace

Kriechi, 2024-01-20:
Please share a g-code file that was generated with a Marlin printer selected in Cura, and also one file with a Duet RRF printer  selected.

Please also upload your cura.log file that contains the log entries of both these files being generated.

Kriechi, 2024-05-26:
Please try installing from source using the next branch, which contains a new commit to address this issue.
https://github.com/Kriechi/Cura-DuetRRFPlugin/tree/next

commit fafa99fb (2024-05-26): refactor thumbnail generation and embedding

closes #52, closes #53
commit 0d9c4dd9 (2024-06-01): refactor thumbnail generation and embedding

closes #52, #53, #54
commit 67a41bee (2024-06-15): refactor thumbnail generation and embedding

closes #52, #53, #54
closed 2024-06-15 by Kriechi