Kriechi/Cura-DuetRRFPlugin #53: Cura 5.7 beta crashes when trying to upload a file via the plugin

oliof, 2024-03-30:
**Describe the bug**
Uploading files via the plugin causes to crash

**To Reproduce**
Steps to reproduce the behavior:
1. configure a printer to use the RepRapFirmware plugin
2. slice
3. hit "Print on <printername>" button
4. Cura crashes, dumping core

**Expected behavior**
Sending files to RRF should work

**Version Information**
 - Operating System: POP!_OS 22.04
 - Cura Version: 5.7-beta1
 - Cura-DuetRRF Plugin Version and Installation: 1.2.10 from Marketplace
 - Duet RepRapFirmware Version: 3.5-rc3
 - 
**Cura Log**

[cura.log](https://github.com/Kriechi/Cura-DuetRRFPlugin/files/14812953/cura.log)

Kriechi, 2024-03-31:
Thanks for the report - unfortunately I don't see an error in the logs related to the Duet-RRF plugin.

I see successfully rendered files and thumbnails from the plugin, e.g., from `green.gcode`, but this seems like you used the "Save to file" button.

I see one attempt using the "Print on" button around timestamp `2024-03-30 19:19:28`, but the logs just end and then 70sec later Cura seems to start-up again -- but there are no error logs or other indications what when wrong. I'm afraid we would need a Cura/UltiMaker dev input for that.

Can you delete (empty) your log file, and try with a clean Cura start-up -- without any other plugins loaded.
Then mark the time when you were able to trigger a crash (i.e., clicked the "print on" button), so we can compare it against the timestamps in the log file.

Can you replicate your issue using the same Cura config and profiles but on a different operating system / computer? POP!_OS sounds like Raspberry Pi - and there might be a Cura or plugin incompatibility in this lesser known OS - but I'm just speculating at this point.

oliof, 2024-03-31:
Hi,
POP!_OS is a Ubuntu derivative, and I am running the standard AppImage. I did save the Gcode file to print after Cura repeatedly crashed. I will provide a clean log file in a short while.

-------- Ursprüngliche Nachricht --------

momenso, 2024-05-02:
I can reproduce and confirm this issue on Ultimaker Cura 5.7.1.
Here is the logs:

`Fatal Python error: Segmentation fault

Current thread 0x000074264ea3b740 (most recent call first):
  File "UM/View/RenderBatch.py", line 309 in _renderItem
  File "UM/View/RenderBatch.py", line 231 in render
  File "cura/PreviewPass.py", line 132 in render
  File "/home/momenso/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFPlugin/thumbnails.py", line 39 in render_scene
  File "/home/momenso/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFPlugin/thumbnails.py", line 120 in generate_thumbnail
  File "/home/momenso/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFPlugin/DuetRRFPlugin.py", line 61 in _embed_thumbnails
  File "UM/Signal.py", line 357 in __performEmit
  File "UM/Signal.py", line 323 in __handleEmitIndirect
  File "UM/Signal.py", line 216 in emit
  File "UM/Signal.py", line 361 in __performEmit
  File "UM/Signal.py", line 323 in __handleEmitIndirect
  File "UM/Signal.py", line 216 in emit
  File "/home/momenso/.local/share/cura/5.7/plugins/DuetRRFPlugin_dev/DuetRRFOutputDevice.py", line 217 in _onFilenameAccepted
  File "UM/Qt/QtApplication.py", line 420 in exec
  File "cura/CuraApplication.py", line 953 in run
  File "cura_app.py", line 243 in <module>`

oliof, 2024-05-14:
I can confirm this still happens with 5.7.1 for me. I haven't been at my printer in a long while, so I still owe you a set of log files.

mariusmuja, 2024-05-26:
Confirming the issue on 5.7.1 as well. As a workaround, the crash doesn't happen if I do "Save to Disk" before.

Kriechi, 2024-05-26:
Please try installing from source using the `next` branch, which contains a new commit to address this issue.
https://github.com/Kriechi/Cura-DuetRRFPlugin/tree/next

I don't know why or where exactly this is crashing, as a `Fatal Python error: Segmentation fault` points to a problem with Cura or one of its OS-level dependencies.

oliof, 2024-06-01:
I can confim that both the workaround in https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/53#issuecomment-2131998727 (save to disk before upload/print/simulate) and disabling thumbnail generation work.I did poke around in the cura directory and actually found some more info about the crash in $HOME/.local/share/cura/stderr.log. I have attached it here.

[cura-5.7-stderr.log](https://github.com/user-attachments/files/15520967/cura-5.7-stderr.log)

oliof, 2024-06-01:
Added a bit more log output and am now pretty sure the thumbnail generation crashes inside `Snapshot.snapshot` for some reason.

from cura.log:

```
2024-06-01 10:13:39,899 - INFO - [MainThread] DuetRRFPlugin.thumbnails.generate_thumbnail [53]: Starting thumbnail generation
2024-06-01 10:13:39,899 - INFO - [MainThread] DuetRRFPlugin.thumbnails.generate_thumbnail [70]: Rendering thumbnail image in sizes: [(48, 48), (240, 240), (320, 320)]
2024-06-01 10:13:39,899 - INFO - [MainThread] DuetRRFPlugin.thumbnails.generate_thumbnail [74]: Rendering thumbnail at size 48 x 48
2024-06-01 10:13:39,900 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2024-06-01 10:13:39,901 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_CuraxmRUBW/share/cura/resources/shaders/overhang.shader]...
2024-06-01 10:13:39,903 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_CuraxmRUBW/share/cura/resources/shaders/transparent_object.shader]...
2024-06-01 10:13:39,904 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_CuraxmRUBW/share/cura/resources/shaders/striped.shader]...
```

oliof, 2024-06-01:
Taking another look at the stderr.log

```
Thread 0x0000733f0fe00640 (most recent call first):
  File "/home/hwa/.local/share/cura/5.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/src/zeroconf/_services/browser.py", line 541 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000733f18c00640 (most recent call first):
  File "selectors.py", line 469 in select
  File "asyncio/base_events.py", line 1860 in _run_once
  File "asyncio/base_events.py", line 600 in run_forever
  File "/home/hwa/.local/share/cura/5.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/src/zeroconf/_core.py", line 494 in _run_loop
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000733f19600640 (most recent call first):
  File "/tmp/.mount_CuraJPDlcJ/share/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap
  ```
  
  I disabled the Octoprint plugin to see if that helps; only marginally since the crash traceback then starts in RemovableDriveOutputDevice, which cannot be disabled via UI. I will look into modifying the AppImage to get that plugin disabled for testing.

Kriechi, 2024-06-01:
These are mostly background threads - and I think not relevant in this case.

The important thread is this one:
```
Current thread 0x0000733f93e3d740 (most recent call first):
  File "UM/View/RenderBatch.py", line 318 in _renderItem
  File "UM/View/RenderBatch.py", line 231 in render
  File "cura/PreviewPass.py", line 132 in render
  File "cura/Snapshot.py", line 194 in snapshot
  File "/home/hwa/.local/share/cura/5.7/plugins/DuetRRFPlugin/thumbnails.py", line 73 in generate_thumbnail
  File "/home/hwa/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFPlugin.py", line 65 in _embed_thumbnails
  File "UM/Signal.py", line 357 in __performEmit
  File "UM/Signal.py", line 323 in __handleEmitIndirect
  File "UM/Signal.py", line 216 in emit
  File "UM/Signal.py", line 361 in __performEmit
  File "UM/Signal.py", line 323 in __handleEmitIndirect
  File "UM/Signal.py", line 216 in emit
  File "/home/hwa/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFOutputDevice.py", line 214 in _onFilenameAccepted
  File "UM/Qt/QtApplication.py", line 420 in exec
  File "cura/CuraApplication.py", line 953 in run
  File "cura_app.py", line 243 in <module>
```

And as you pointed out, it crashes in the Snapshot.snapshot function, which is called by the plugin code:
https://github.com/Kriechi/Cura-DuetRRFPlugin/blob/fafa99fb11ee15647078a694940f33584812ddc8/thumbnails.py#L73

Looks like a rather straight forward function call to me - so I suspect an issue with Cura or underlying libraries. I suggest opening a new issue with Cura - there have been a few similar ones in the past: https://github.com/search?q=repo%3AUltimaker%2FCura+snapshot.snapshot&type=issues

Kriechi, 2024-06-01:
I also pushed a new commit with a suggested fix from one of these issues:
https://github.com/Kriechi/Cura-DuetRRFPlugin/commit/9b79f98bc8b312eaf6e8da577450afea4953f1d9

Please test the latest version from the branch, or make the changes in the mentioned 3 lines of code.

oliof, 2024-06-01:
Updated next branch, the crash still happens, _call_on_qt_thread_wrapper now part of the trace.

The workaround with saving to disk first still applies.

```
Current thread 0x000076a34c6b3740 (most recent call first):
  File "UM/View/RenderBatch.py", line 318 in _renderItem
  File "UM/View/RenderBatch.py", line 231 in render
  File "cura/PreviewPass.py", line 132 in render
  File "cura/Snapshot.py", line 194 in snapshot
  File "/home/hwa/.local/share/cura/5.7/plugins/DuetRRFPlugin/thumbnails.py", line 76 in generate_thumbnail
  File "cura/Utils/Threading.py", line 31 in _call_on_qt_thread_wrapper
  File "/home/hwa/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFPlugin.py", line 65 in _embed_thumbnails
  File "UM/Signal.py", line 357 in __performEmit
  File "UM/Signal.py", line 323 in __handleEmitIndirect
  File "UM/Signal.py", line 216 in emit
  File "UM/Signal.py", line 361 in __performEmit
  File "UM/Signal.py", line 323 in __handleEmitIndirect
  File "UM/Signal.py", line 216 in emit
  File "/home/hwa/.local/share/cura/5.7/plugins/DuetRRFPlugin/DuetRRFOutputDevice.py", line 214 in _onFilenameAccepted
  File "UM/Qt/QtApplication.py", line 420 in exec
  File "cura/CuraApplication.py", line 953 in run
  File "cura_app.py", line 243 in <module>
```

oliof, 2024-06-01:
Looking at Snapshot.py and RenderBatch.py between 5.6 and 5.7, there haven't been many changes: Snapshot.py got a slight refactor and RenderBatch.py is unchanged. So it's most likely underlying Qt or GL changes that cause the issue here. I will take a break and investigate a bit further later.

Kriechi, 2024-06-01:
@oliof thanks - I've pushed another commit, moving the `call_on_qt_thread` to a different place (earlier in the code execution path). Please try this.

oliof, 2024-06-01:
no meaningful improvement. I have a trivial STL that doesn't always crash (basically a washer), but even a simple plate with four holes makes things go sideways.

Kriechi, 2024-06-26:
I've now published v1.2.11 which includes an option to enable/disable thumbnail creation and also to select which sizes to generate.

If anyone still has issues with crashes like discussed here, please directly open a new *bug report* with Cura:
https://github.com/Ultimaker/Cura/issues/new/choose to help us track down the details.

commit fafa99fb (2024-05-26): refactor thumbnail generation and embedding

closes #52, closes #53
closed 2025-01-11 by Kriechi