Kriechi/Cura-DuetRRFPlugin #54: Feature request: selectable thumbnail sizes?

DropbearNinja, 2024-05-12:
This is (hopefully) a very small feature request: I don't need or use all produced thumbnail images, but I do like occasionally look at the g-code in a file, but having 4 thumbnails (and the 256 which is really long) takes up heaps of space at the front of the file. Could we have something like checkboxes in the config box to enable/disable certain thumbnail sizes?

Kriechi, 2024-05-26:
Please try installing from source using the next branch, which contains a new commit to address this issue.
https://github.com/Kriechi/Cura-DuetRRFPlugin/tree/next

You can enable/disable, and also define which thumbnail sizes to generate from the `Connect Duet RepRapFirmware` settings dialog in the `Manage Printers` Cura menu.

Kriechi, 2024-06-26:
fixed in v1.2.11

closed 2024-06-26 by Kriechi