Kriechi/Cura-DuetRRFPlugin #55: [1.2.11 approval process] Snapshots working on 5.0.0, fail on 5.7.0

rburema, 2024-06-25:
Communicating here gives me a bit more leeway than 'accept/reject' + 1 single line of feedback.

I used the same printer (maukcc/cartesio) on both versions, pretended to connect so I can test the snaphot-functionality. Which seems to work on 5.0, but I get this in the logs in 5.7 (10x per tried format):

~~~
2024-06-25 10:25:21,659 - INFO - [MainThread] cura.Snapshot.snapshot [204]: Trying to get the snapshot again.
2024-06-25 10:25:21,664 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [252]: Changing perspective mode to orthographic.
2024-06-25 10:25:21,666 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\overhang.shader]...
2024-06-25 10:25:21,681 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\transparent_object.shader]...
2024-06-25 10:25:21,692 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\striped.shader]...
2024-06-25 10:25:21,744 - INFO - [MainThread] cura.Snapshot.snapshot [204]: Trying to get the snapshot again.
2024-06-25 10:25:21,747 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [252]: Changing perspective mode to orthographic.
2024-06-25 10:25:21,749 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\overhang.shader]...
2024-06-25 10:25:21,764 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\transparent_object.shader]...
2024-06-25 10:25:21,775 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.7.0\share\cura\resources\shaders\striped.shader]...
2024-06-25 10:25:21,829 - WARNING - [MainThread] cura.Snapshot.snapshot [200]: Failed to crop the snapshot even after 10 attempts!
2024-06-25 10:25:21,836 - DEBUG - [MainThread] DuetRRFPlugin.thumbnails.generate_thumbnail [75]: Skipping failed 240x240 thumbnail.
~~~

Kriechi, 2024-06-25:
Thanks for jumping in here to discuss!

For your printer "maukcc/cartesio", did you configure the DuetRRF plugin in any way, or did you just install it, and use an already existing printer from your Cura profile?

For "un-configured" printers, this plugin should do nothing - and now I'm wondering if the default values or default config are incorrect? Would you mind sharing all log lines containing `DuetRRFPlugin ` please?

Do you have a `[duetrrf]` section in your `cura.cfg`?

rburema, 2024-06-26:
I wanted to get a cleaner log, so I just cleared everything out and tried again. That seemed to work. It must've been something in my config.

Sorry for the inconvenience!

For future reference, should it come up, in both cases:

- I added a new printer to Cura (I wanted to test it on a non-UM machine, so I just picked an arbitrary 3rd-party one) -- so not one we've got here physically.
- Went to the printer config window to click the 'Connect Duet RepRap Firmware' button.
- Didn't fill anything else in (but observed that the 'generate snapshot' checkmark was checked in both cases), and clicked 'save config'.
- Slice & save to file, inspect gcode output.

closed 2024-06-26 by rburema