Kriechi/Cura-DuetRRFPlugin #56: No upload with Duet RRF 3.6.0-beta.2 (internal server error)

NikAWing, 2024-12-05:
Hello!

Clicking on upload to printer results in "internal server error" and no upload happens.
Uploading gcode via Web GUI works.
It must be related to RRF 3.6.0-beta.2, with beta.1 I did not have that problem.

**Expected behavior**
Well, a perfect upload to the printer :)

**Version Information**
Cura 5.9.0 on Win11

**Cura Log** excerpt:

2024-12-05 19:16:53,859 - DEBUG - [MainThread] DuetRRFPlugin.helpers.serializing_scene_to_gcode [13]: Serializing gcode...
2024-12-05 19:16:53,969 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._onFilenameAccepted [233]: Connecting...
2024-12-05 19:16:54,381 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [394]: request[83eec2f3][get][PyQt6.QtCore.QUrl('http://192.168.1.12/rr_connect?password=&time=2024-12-05T19%3A16%3A53')][timeout=None][no-data] got an QNetworkReplyError Error transferring http://192.168.1.12/rr_connect?password=&time=2024-12-05T19%3A16%3A53 - server replied: Internal Server Error. The server returned: b'"expected start of object"'
2024-12-05 19:16:54,390 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [241]: rr_connect failed with error NetworkError.InternalServerError
2024-12-05 19:16:54,390 - ERROR - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._onNetworkError [534]: <NetworkError.InternalServerError: 401>

FractalEngineer, 2024-12-05:
Hah I came to post about the same issue here 😅

`There was a network error: NetworkError.InternalServerError Error transferring http://192.168.0.30/rr_connect?password=&time=2024-12-06T03%3A36%3A00 - server replied: Internal Server Error`

Kriechi, 2024-12-06:
`server replied: Internal Server Error.` sounds like this is best reported to the RRF forum - have you done that already?
Please follow the instructions in https://forum.duet3d.com/topic/36941/version-3-6-0-beta-2-now-available

I don't see any relevant API changes in the changelog. @dc42 any pointers on what is happening here?

NikAWing, 2024-12-08:
Hey Kriechi,
I did not yet report it in the RRF forum, there was not much time yet and I thought it's rather some kind of change in RRF that needs to be addressed here :)
I will test again and then go to RRF forum.

dc42, 2024-12-09:
Is the Duet running in standalone mode or SBC mode (i.e. with attached single board computer)?

NikAWing, 2024-12-09:
it's in SBC mode with a raspberry

(btw, this has nothing to do with the problem, but I always hat an old input shaper plugin listed. I think v3.4.x ... I could never uninstall it, but I suddenly was able to do this after the recent beta update, yay)

NikAWing, 2024-12-15:
ok, just tried it again.
Booted the printer, connected via browser and Duet web control told me that not all plugins were loaded. I had to start them manually (DuetPi management plugin, height map, input shaping and g-code viewer). This is the 2nd time since I updated to 3.6.0ß2

And uploading gcode from Cura with DuetRRF plugin did not work, same error as above.
Should I go over to RRF forum and open an issue?

Kriechi, 2024-12-15:
yes, please go to the forum and follow the instructions in https://forum.duet3d.com/topic/36941/version-3-6-0-beta-2-now-available to open a new issue related to the RRF beta. I'll keep monitoring.

Kriechi, 2025-01-11:
Should be fixed in the next beta release of RRF/DSF:
https://forum.duet3d.com/topic/36941/version-3-6-0-beta-2-now-available/14?_=1736601869457

closed 2025-01-11 by Kriechi